platformio/platformio-vscode-ide #604: Extension causes high cpu load

SashaFed, 2019-02-21:
- Issue Type: `Performance`
- Extension Name: `platformio-ide`
- Extension Version: `1.6.0`
- OS Version: `Windows_NT ia32 6.1.7601`
- VSCode version: `1.31.1`

:warning: Make sure to **attach** this file from your *home*-directory: `C:\Users\Aleksandr\platformio.platformio-ide-unresponsive.cpuprofile.txt` :warning:

Find more details here: https://github.com/Microsoft/vscode/wiki/Explain:-extension-causes-high-cpu-load

SashaFed, 2019-02-22:
[platformio.platformio-ide-unresponsive.cpuprofile.txt](https://github.com/platformio/platformio-vscode-ide/files/2894012/platformio.platformio-ide-unresponsive.cpuprofile.txt)

SashaFed, 2019-02-22:
https://github.com/platformio/platformio-vscode-ide/issues/586     ))))))

SashaFed, 2019-02-22:
https://github.com/platformio/platformio-vscode-ide/issues/586

ivankravets, 2019-02-22:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/478

closed 2019-02-22 by ivankravets