platformio/platformio-vscode-ide #606: FEATURE: Copy and paste values from IDE Plugin

exocode, 2019-02-23:
As a developer I am lazy. I wanna be able to mark, copy and paste definitions as well examples.
When I open the PlatformIO plugin in VSCode and choose a library and an example of it, I am not able to Copy this code. I can mark it, but copy/pasting is not working.

I have to visit the PlatformIO website and have to do it from there.

ivankravets, 2019-03-04:
What is your OS?

exocode, 2019-03-04:
MacOS 10.13.6 (17G4015)

ketan, 2019-03-21:
Same here. Unable to copy/paste from platformio IDE in vscode.

exocode, 2019-04-19:
Something new on that task?

I tried today copy/pasting in "ATOM" editor, it is working. 
**Still not working with VSCode.**

Please keep this issue/feature in focus. I "think" it is only a setting/config problem.

ivankravets, 2019-04-19:
This is a VSCode bug, see https://github.com/Microsoft/vscode/issues/65452

@bpasero @mjbvz do we have any workarounds?

ivankravets, 2019-04-19:
A temporary solution is to open PIO Home in a standalone browser. See screenshot below.
[image]

ivankravets, 2019-12-17:
A temporary workaround is to use `VSCode > Menu > Edit > Copy/Paste`

closed 2020-06-11 by ivankravets

ranger81, 2020-06-12:
Installing the new PIO home 3.2.3 does not solve the issue for me.

ivankravets, 2020-06-12:
You also need the latest PlatformIO for VSCode IDE 2.0 https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta/14097

danielo515, 2021-11-12:
The mentioned issue is closed as wontfix and this is the workaround they propose for the iframe:

https://github.com/microsoft/vscode/issues/65452#issuecomment-586485815

commit 128cb8d9 (2020-06-11): Fixed a bug when hotkeys in PIO Home does not work macOS // Resolve #606