platformio/platformio-vscode-ide #618: [bug] Installation Manager: Could not install packages due to an EnvironmentError: Could not find a suitable TLS CA certificate bundle

snowyu, 2019-03-02:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.31.1
PIO IDE: v1.6.0
System: Linux, 4.18.0-15-generic, x64

# Exception
```
Error: PIP Core: Could not install packages due to an EnvironmentError: Could not find a suitable TLS CA certificate bundle, invalid path: ~/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/certifi/cacert.pem

	at f.value.i.runCommand (/home/riceball/.vscode/extensions/platformio.platformio-ide-1.6.0/node_modules/platformio-node-helpers/dist/index.js:1:26217)
	at ChildProcess.h (/home/riceball/.vscode/extensions/platformio.platformio-ide-1.6.0/node_modules/platformio-node-helpers/dist/index.js:1:5916)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.EventEmitter.emit (domain.js:442:20)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

## Workaround

Uninstall and reinstall it.

ivankravets, 2019-03-04:
Have you resolved this issue?

snowyu, 2019-03-06:
My workaround way is : uninstall it in vscode and install it again.

But I do not know why first install is failed.

ivankravets, 2019-03-06:
Thanks for the solution!

closed 2019-03-06 by ivankravets