platformio/platformio-vscode-ide #623: Debugger breakpoints are not activated on new debug session start

me21, 2019-03-03:
### Configuration

**Operating system**: Windows 10 x64 v1607 LTSB Russian

**PlatformIO Version** (`platformio --version`): 3.6.4

### Description of problem
The breakpoints I set before I start debug session are not automatically activated when I start a new debug session.

#### Steps to Reproduce

1. Open some "hello world" example.
2. Set a breakpoint somewhere so that it will be hit during program execution.
3. Debug the project.
4. After the execution stops at the beginning of the `main` function, press "Continue" (F5).

### Actual Results
The breakpoint is not hit.

### Expected Results
The breakpoint should be hit.

### Additional info
Workaround: set the breakpoints **after** you start the debug session and the execution stops at the beginning of the `main` function.

jeremypoulter, 2019-03-12:
Same is happening for me:

**Operating system**: Ubuntu 18.04.2 LTS, Linux 4.15.0-46-generic #49-Ubuntu SMP Wed Feb 6 09:33:07 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

**PlatformIO Version** (`platformio --version`): 3.6.4
**Platform**: Nordic nRF52 (Adafruit BSP)
**Board**: Custom
**Framework**: Arduino
**Debugger**: J-Link

ivankravets, 2019-05-06:
@me21 what is your `platformio.ini`?

ivankravets, 2019-05-28:
Please reopen if you still need help

phaseOne, 2019-05-28:
Could we please reopen this? I experience the same issue with the Arduino framework, and a J-Link. I can send along my platformio.ini, if that is helpful.

ivankravets, 2019-05-28:
@phaseOne Sorry, I can't reproduce it. Could provide more details or a simple project to reproduce this issue? Thanks!

phaseOne, 2019-05-28:
I'll compile a more detailed report, and get back to you by next week. Any recommendations for what I should inspect (e.g. logfiles, console, etc.) while observing the issue?

It's the same behavior @me21 reported:
1. start a debug session with the VSCode Debug Panel's green "run" button
2. set some breakpoints
3. end the debug session, or restart it
4. debugger doesn't hit the breakpoints

After restarting the session, the debugger will only hit the breakpoints if I manually untick and retick the checkboxes in the breakpoints list.

S2Doc, 2019-06-18:
I also have same problem.  This makes debugging very difficult.

Ivan, you asked about platform.ini.  This is mine:

[env:adafruit_feather_m4]
platform = atmelsam
board = adafruit_feather_m4
framework = arduino
debug_tool = jlink
upload_port = COM[3]
upload_protocol = jlink
monitor_speed = 115200
lib_deps = WiFi101, Adafruit_MQTT, pgmStrToRAM

ivankravets, 2019-08-31:
Hi all, 

I need your help to reproduce this issue. Please do the next:

1) Open [/platformio/commands/debug/client.py](https://github.com/platformio/platformio-core/blob/develop/platformio/commands/debug/client.py#L37) and edit line 37. Set a path to log file. PlatformIO Core will write here whole communication between GDB Client and debugging server. For example, `LOG_FILE = "/tmp/piodebug.txt"` or `LOG_FILE = "C:/tmp/piodebug.txt"`.

How to find where this file is located on your machine? Please open PlatformIO IDE Terminal and type:
```
python -c "from platformio.commands import debug; print(debug)"
```

2) Attached this file here or share on https://hastebin.com. You can cleanup your username and other sensitive information. What I actually need that is a sequence when VSCode sends `-break-insert` command. It should be AFTER `~"PlatformIO: Initialization completed\n"`. 

----------------
So, after enabling logging, please do the next:

a) Open VSCode, set breakpoints where you want, or, reproduce your workflow when this issue happens
b) Close VSCode
c) Open VSCode
d) Start debugging
c) Share this debug log somewhere.

Thanks!

jeremypoulter, 2019-09-04:
Here you go: [piodebug-2019-09-04.txt](https://github.com/platformio/platformio-vscode-ide/files/3574396/piodebug-2019-09-04.txt)

FYI this was on a ESP Wrover kit using the ESP32 arduino blink example

jeremypoulter, 2019-09-04:
Also here is the progeck dir, including object files, etc just in case it helps
[arduino-blink.zip](https://github.com/platformio/platformio-vscode-ide/files/3574403/arduino-blink.zip)

S2Doc, 2019-09-04:
@ivankravets Here is the log file generated per your instructions:  
[piodebug.txt](https://github.com/platformio/platformio-vscode-ide/files/3575110/piodebug.txt)
Here is the complete project folder:  
[M4 ASCIItest.zip](https://github.com/platformio/platformio-vscode-ide/files/3575120/M4.ASCIItest.zip)
Note that the Monitor is used.

There is a breakpoint set at line 52.  This works when first placed, but is ignored after starting another Debug session or after quitting VSCode and restarting it, then starting a new Debug session.

ivankravets, 2019-09-04:
Logs look OK. Could you try to list GDB breakpoints when session is started? In VSCode Debugging Console there is "GDB Console", type
```
info b
```

So, it should list declared breakpoints in this session.

S2Doc, 2019-09-04:
I started up the project again in Debug mode.  The breakpoint at line 52 is ignored.  Here is the output from the GDB console:
```
info b
{"token":20,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
No breakpoints or watchpoints.
```

ivankravets, 2019-09-04:
Thanks! Could you try adding 
```
[env:yourenv]
board = ...
debug_init_break =
```
So, just keep it empty. Does it work better now?

S2Doc, 2019-09-04:
YES!

That works.   It now preserves and observes breakpoints between Debug sessions and also when I shutdown and restart VSCode.   It no longer stops at init() when satarting a session; instead it stops in the reset handler function.

Thank you, Ivan!   Will you be incorporating this into a future release?

ivankravets, 2019-09-04:
Yes, I see where is problem. I'll back here soon. Thank you so much for provided info!

S2Doc, 2019-09-04:
Thank you for helping with the problem!

closed 2019-10-11 by ivankravets

ivankravets, 2019-10-11:
Hi all,

Could you try the upcoming PlatformIO IDE for VSCode 1.9.0. See instruction in "Solution" section here https://github.com/platformio/platformio-vscode-ide/issues/959

Does it work for you?

ivankravets, 2019-10-11:
1.9.0 is out! https://github.com/platformio/platformio-vscode-ide/releases/tag/v1.9.0

It also fixes an issue for ESP32 when breakpoint is set in project.

cstrahan, 2023-02-02:
@ivankravets Could you point me at the the commit that fixes this? https://github.com/platformio/platformio-vscode-ide/commit/117491250035060773bc25061a6fc70081b65ec7 bumps one of the dependency versions, but I'm curious about what the underlying fix looked like, as I'm also running into this issue and hoping to glean some info.

jpmeijers, 2023-04-24:
Also affected here using stlink on ststm32 platform. Adding `debug_init_break =` (empty) to the platformio.ini doesn't help and actually makes the problem worse.

keenedge, 2023-10-10:
I'm still seeing the same problem.   Intermittently stops at breakpoints.  Tried With and Without debug_init_break =

I Get a messages say in "line 22 does not exist in main.cpp" when I  clear have a line 22 and my breakpoint is set there

Windows 11
Platformio: Core 6.1.11·Home 3.4.4
SEEED XIAO ESP32-S3

commit 11749125 (2019-10-11): Fixed an issue when debugger breakpoints are not activated on new debug session start // Resolve #623