platformio/platformio-vscode-ide #624: Multiple instances of Aceinna Extension Home Page loaded at the same time

AndreyBondarev, 2019-03-04:
Whenever I removing project from workspace and open another project - Aceinna home webpage gets reloaded even if it was previously opened.
[image]

ivankravets, 2019-07-15:
Please re-test with https://github.com/platformio/platformio-vscode-ide/releases/tag/v1.8.0-beta.1

AndreyBondarev, 2019-07-15:
Still there is another instance in case when I choose to “remove folder from workspace” and checkbox “Show at startup” checked.
The sequence is:

1.       Open project in the existing workspace – homepage appears.

2.       Remove folder from workspace – another instance of homepage appears

Andrey

closed 2019-07-16 by ivankravets

AndreyBondarev, 2019-07-17:
Hi Ivan

For some reason now sometimes I see “PIO Home” Tab name instead of “Aceinna Home” tab when opening new project in arbitrary directory.

[cid:image001.jpg@01D53C9E.D1B1FDE0]

Andrey

Closed #624<https://github.com/platformio/platformio-vscode-ide/issues/624> via a47565c<https://github.com/platformio/platformio-vscode-ide/commit/a47565c4ab087229a36a458701237d718200498d>.

ivankravets, 2019-07-17:
@AndreyBondarev please file a new issue here https://github.com/platformio/platformio-vscode-ide/issues

I'll work on it a little bit later. Thanks!

commit dd7b1454 (2019-07-15): Fixed an issue with multiple instances of PIO Home when opening new project // Resolve #624
commit a47565c4 (2019-07-16): Close PIO Home when workspace folders are changed // Resolve #624