platformio/platformio-vscode-ide #631: Rebuild IntelliSense Index randomly giving permission errors

kobuki, 2019-03-09:
When I'm hitting the `Rebuild IntelliSense Index` menu on the VSCode PIO GUI, I randomly get the exception below. Sometimes it succeeds, but most of the time it fails and I have this error. It's a basic Arduino project with a few lines of test code.

Win7 x64, VSCode 1.32.1, PIO Home 2.0.0, Core 3.6.5

```
> Executing task: c:\Users\user\.platformio\penv\Scripts\platformio.exe init --ide vscode <

The current working directory d:\Work\arduino\piotest1 will be used for the project.

The next files/directories have been created in d:\Work\arduino\piotest1
include - Put project header files here
lib - Put here project specific (private) libraries
src - Put project source files here
platformio.ini - Project Configuration File
Error: Traceback (most recent call last):
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\user\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\user\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\user\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\user\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\user\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\user\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\commands\init.py", line 101, in cli
    pg = ProjectGenerator(project_dir, ide, env_name)
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 36, in __init__
    self._gather_tplvars()
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 138, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 76, in get_project_build_data
    raise result.exception
IOError: [Errno 13] Permission denied: 'c:\\Users\\user\\.platformio\\packages.lock'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

ivankravets, 2019-04-05:
Please check that antivirus tools don't block PlatformIO Core. Try to remove `c:\\Users\\user\\.platformio` folder and restart VSCode.

closed 2019-04-05 by ivankravets

kobuki, 2019-04-05:
When I stopped my AV, the problem still persisted. Did a .patformio remove, helped for a while then the problem reappeared. Past several minor versions it didn't come up again. It also helped with a problematic version to create the packages.lock file by hand. Weird. If/when I encounter the issue again, I'll post more info.