platformio/platformio-vscode-ide #634: Example setup error

clplaneguy, 2019-03-11:
My project is to use the Makeblock Ultimate 2.0 robot kit to create a balancing robot that is controlled wirelessly by a Nintendo Wii nunchuck starting from the firmware.  The 3,000 lines of code is so complicated and involved that it is hard to understand.  I have progressed to the point that I have what I think of as a vibration(timing) problem.   I am ready to move on from the Arduino IDE to something that can give me more information and help. 

I installed Oricle VM VirtialBox and ran into a problem with Java.  So, I installed Visual Studio Code and the Arduino extension.  But with Blink I cannot get past a "Serial.print" error.  Literature says that the PlatformIO extension is better than the Arduino extension.  So, I installed it and am following the instructions in the User guide.  
https://docs.platformio.org/en/latest/ide/vscode.html#user-guide 

But, I get the following error.  **I do not see a solution.**

 ```
 #include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

> #include errors detected. Please update your includePath. IntelliSense features for this translation unit (C:\Users\CL Plane Guy 3\Documents\PlatformIO\Projects\PlatformIO first project\src\main.cpp) will be provided by the Tag Parser.

> #include errors detected. Please update your includePath. IntelliSense features for this translation unit (C:\Users\CL Plane Guy 3\Documents\PlatformIO\Projects\PlatformIO first project\src\main.cpp) will be provided by the Tag Parser.
 
> cannot open source file "stdlib.h" (dependency of "Arduino.h")

When I go to the includePath, it says not to edit(?).

**What do I do?**

> Commit: 05f146c7a8f7f78e80261aa3b2a2e642586f9eb3
> Date: 2019-03-08T00:51:49.278Z
> Electron: 3.1.3
> Chrome: 66.0.3359.181
> Node.js: 10.2.0
> V8: 6.6.346.32
> OS: Windows_NT x64 6.1.7601
>

ivankravets, 2019-04-05:
Could you remove `.vscode` folder from a project and restart VSCode?

ivankravets, 2019-05-28:
Please reopen if you still need help

closed 2019-05-28 by ivankravets

vance, 2020-03-08:
Having the same problem... If i remove the .vscode, then none of the arduino built-ins work.

ivankravets, 2020-03-09:
@vance what is your os?