platformio/platformio-vscode-ide #639: checking platformio core installation infinitely loop

jenokizm, 2019-03-14:
hi i launched wednesday a month later last work. Your plugin has detected an update but status "**checking platformio core installation**" infinitely. I waited for the result an hour. This installation is the case with this I can not admire the plug-ins at this moment. Then I closed and restarted **Visual Studio Code** but the installation is still so spinning. I waited another hour but to no avail. What should I do?

ivankravets, 2019-04-05:
Please remove `C:/Users/%Name%/.platformio/penv` folder and restart VSCode.

closed 2019-04-05 by ivankravets

zhoushenglin, 2020-02-21:
Please try to kill some "python.exe".

sinnfeinn, 2020-02-25:
On mac closing python processes via activity monitor helped to end the loop.

evanic14, 2022-08-26:
This totally worked for me! Thanks a lot!