platformio/platformio-vscode-ide #661: c_cpp_properties: compilerPath syntax

KarateBrot, 2019-03-28:
# Problem
Referring to [this issue](https://github.com/Microsoft/vscode-cpptools/issues/3357) I opened on the [vscode-cpptools](https://github.com/Microsoft/vscode-cpptools) github page, **paths with whitespaces in it will fail in the compilerPath** entry of `c_cpp_properties.json`.

### Failing example:
```javascript
"compilerPath": "C:/Users/Jan Post/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-gcc.exe -mlongcalls -mtext-section-literals"
```

### More readable format:
```
"compilerPath": "PATH WITH WHITESPACES -args"
```

# Solution
When `c_cpp_properties.json` gets created, there are `\"` brackets missing in the `compilerPath`. Since `compilerPath` gets passed with args, the according file path needs to be enclosed in `\"` brackets. Otherwise file paths with whitespaces in it will fail.

### Working example:
```javascript
"compilerPath": "\"C:/Users/Jan Post/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-gcc.exe\" -mlongcalls -mtext-section-literals"
```

### More readable format:
```
"compilerPath": "  \"  PATH WITH WHITESPACES  \" -args"
```

ivankravets, 2019-04-01:
Fixed in the latest PIO Core dev/version. Please re-test with setting [platformio-ide.useDevelopmentPIOCore = true](http://docs.platformio.org/en/latest/ide/vscode.html#platformio-ide-usebuiltinpiocore)

closed 2019-04-01 by ivankravets

KarateBrot, 2019-04-01:
Fixed!

Thank you!