platformio/platformio-vscode-ide #672: Installation Manager

h4koo, 2019-04-03:
[image]
# Description of problem
Hi, it seems like the VS extension for PlatformIO is not being able to download the virtual environment file. 

I have checked https://github.com/platformio/platformio-vscode-ide/issues/154, but it seems like the issue was solved on version 1.5, I'm on version 1.6

Please let me know if there is anything I can do to help

[image]

[image]

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.32.3
PIO IDE: v1.6.0
System: Windows_NT, 10.0.16299, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Failed to download file https://files.pythonhosted.org/packages/4e/8b/75469c270ac544265f0020aa7c4ea925c5284b23e445cf3aa8b99f662690/virtualenv-16.1.0.tar.gz: Error: unable to get local issuer certificate
	at t.value (C:\Users\JorgeAgueroZamora\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:25327)
```

[image]

[image]

ivankravets, 2019-04-05:
Please read the whole error which says `Error: unable to get local issuer certificate...`.  It looks that VSCode can't download virtualenv package.

Do you use proxy?

h4koo, 2019-04-07:
Hi Ivan,

I don't think so. 

I was able to get around this by adding this domains to the `%APPDATA%\pip\pip.ini ` file

```
[global]
trusted-host=
	pypi.python.org
	pypi.org
	files.pythonhosted.org
```

but it seems like the problem with the certificates persists since now I'm not bale to load the PIOHome in VSCode

When I run `pio home`
```

Error: Error: Please read http://bit.ly/package-manager-ioerror Error: Could not install 'tool-pioplus' with version requirements '^2.0.0' for your system 'windows_amd64'.  Please try this solution -> http://bit.ly/faq-package-manager      at P.Promise (C:\Users\JorgeAgueroZamora\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:14992)     at ChildProcess.h (C:\Users\JorgeAgueroZamora\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:5916)     at ChildProcess.emit (events.js:182:13)     at ChildProcess.cp.emit (C:\Users\JorgeAgueroZamora\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\cross-spawn\lib\enoent.js:34:29)     at maybeClose (internal/child_process.js:961:16)     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
onDidNotificationChange @ notificationsAlerts.ts:40
console.ts:134 [Extension Host] Error: Error: Please read http://bit.ly/package-manager-ioerror
Error: Could not install 'tool-pioplus' with version requirements '^2.0.0' for your system 'windows_amd64'.

Please try this solution -> http://bit.ly/faq-package-manager

    at P.Promise (C:\Users\JorgeAgueroZamora\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:14992)
    at ChildProcess.h (C:\Users\JorgeAgueroZamora\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:5916)
    at ChildProcess.emit (events.js:182:13)
    at ChildProcess.cp.emit (C:\Users\JorgeAgueroZamora\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:961:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)

t.log @ console.ts:134
log.ts:173   ERR unable to get local issuer certificate: Error: unable to get local issuer certificate
    at TLSSocket.onConnectSecure (_tls_wrap.js:1055:34)
    at TLSSocket.emit (events.js:182:13)
    at TLSSocket._finishInit (_tls_wrap.js:635:8)

3log.ts:173   ERR EPERM: operation not permitted, open 'C:\Users\JORGEA~1\AppData\Local\Temp\vscode-update-system-x64\CodeSetup-stable-0dd516dd412d42323fc3464531b1c715d51c4c1a.exe.tmp': Error: EPERM: operation not permitted, open 'C:\Users\JORGEA~1\AppData\Local\Temp\vscode-update-system-x64\CodeSetup-stable-0dd516dd412d42323fc3464531b1c715d51c4c1a.exe.tmp'
marketplace.visualstudio.com/_apis/public/gallery/extensionquery:1 Failed to load resource: net::ERR_NETWORK_IO_SUSPENDED
log.ts:173   ERR Cannot read property 'selection' of undefined: TypeError: Cannot read property 'selection' of undefined
    at CodeActionProvider.getInitializeFromCtorCommand (C:\Users\JorgeAgueroZamora\.vscode\extensions\jchannon.csharpextensions-1.3.0\out\src\codeActionProvider.js:165:33)
    at addInitalizeFromCtor 
```

ivankravets, 2019-04-08:
The error is reported by Node.JS. What is your Python 2 version?

h4koo, 2019-04-29:
Hi Ivan,

Sorry for the late reply, I had both 2.75 and python 3 versions installed. I'm on my work computer so I'm guessing that may be it. 

Let me know if you want to keep troubleshooting this, or I believe this can be closed.

Thank you very much for your help.

ivankravets, 2019-04-29:
Pleased uninstall Python 2.7.5 and install > 2.7.9. I recommend the latest Python 2.7.

closed 2019-04-29 by ivankravets