platformio/platformio-vscode-ide #675: #include <> errors with the new update

GeorgeFlorian, 2019-04-06:
- VSCode Version: version 1.33
- OS Version: Linux Mint 19.1

Steps to Reproduce:

1. Install the new 1.33 version.

## Issue:

I've never configured includePath or IntelliSense, and never had issues with `#include <>`
I've just upgraded to the newest version of the VS Code (1.33) and I went to compile a random project.

I noticed that it compiles without any problems, but all my `#include <>` are underlined with a green wobbly line. Hovering over the first `#include <>` says: 

```#include errors detected. Please update your includePath. IntelliSense features for this translation unit (/home/george/Documents/PlatformIO/Projects/LoginPage_try5/src/main.cpp) will be provided by the Tag Parser.
```

Hovering over the next few `#include <>` and they all say: `cannot open source file "libraryname.h"` where `libraryname` corresponds to their library names, respectively.

What do you think the issues is and how can I fix it ?

Thank you !

I've also posted on Microsoft vscode's github and they told me to piss off. 
https://github.com/Microsoft/vscode/issues/71784

ivankravets, 2019-04-16:
Please remove `.vscode` folder from a project and restart VSCode.

closed 2019-04-16 by ivankravets

ClearSkySpace, 2019-06-02:
Just had this same issue again after updating to VSCode 1.34.
> Please remove `.vscode` folder from a project and restart VSCode.

That fixed it. Thanks!

_For Google: `#include errors detected. Please update your includePath. IntelliSense features for this translation unit will be provided by the Tag Parser.`
`platrormio "cannot open source file \"Arduino.h\`_

r0oland, 2020-01-19:
for anyone else out there struggling with this...tried to remove the .vscode folder but it made no difference. I had to also remove all workspace folders except the current project folder and then removing the .vscode folder worked!