platformio/platformio-vscode-ide #695: Start PIO Home Server

liuxinge, 2019-04-16:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.33.1
PIO IDE: v1.6.0
System: Windows_NT, 6.1.7601, x64

# Exception
```
Error: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 18, in load_module
  File "<string>", line 8, in <module>
ImportError: No module named serial

    at P.Promise (C:\Users\lukeliu\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:177:28)
    at ChildProcess.h (C:\Users\lukeliu\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\misc.js:167:5)
    at ChildProcess.emit (events.js:182:13)
    at ChildProcess.cp.emit (C:\Users\lukeliu\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:961:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

liuxinge, 2019-04-16:
pyserial 3.4 have been installed.

ivankravets, 2019-04-16:
Could you reproduce this issue after VSCode restart?

liuxinge, 2019-04-17:
> Could you reproduce this issue after VSCode restart?

Yes, I have updated both VScode and PlatformIo to latest version, and have restarted VScode couple of time, it keep on displaying 'loading........', and give out the same error: 

# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.33.1
PIO IDE: v1.7.0
System: Windows_NT, 6.1.7601, x64

# Exception
```
Error: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 18, in load_module
  File "<string>", line 8, in <module>
ImportError: No module named serial

    at P.Promise (C:\Users\lukeliu\.vscode\extensions\platformio.platformio-ide-1.7.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:177:28)
    at ChildProcess.h (C:\Users\lukeliu\.vscode\extensions\platformio.platformio-ide-1.7.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\misc.js:167:5)
    at ChildProcess.emit (events.js:182:13)
    at ChildProcess.cp.emit (C:\Users\lukeliu\.vscode\extensions\platformio.platformio-ide-1.7.0\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:961:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

ivankravets, 2019-04-17:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

liuxinge, 2019-04-17:
> Please provide an output from system Terminal `cmd`
> 
> ```
> echo %PATH%
[path.txt](https://github.com/platformio/platformio-vscode-ide/files/3091773/path.txt)

> echo %PYTHONPATH%
C:\Users\lukeliu\AppData\Local\anaconda3\Lib\site-packages\pyserial-3.4.dist-info;C:\Users\lukeliu\AppData\Local\anaconda3\Lib\site-packages 

> python --version
python 2.7.14
> pip --version
pip 18.1 
> pip list
[piplist.txt](https://github.com/platformio/platformio-vscode-ide/files/3091777/piplist.txt)

Thanks!

ivankravets, 2019-05-28:
Did you resolve this issue?

closed 2019-07-13 by ivankravets