platformio/platformio-vscode-ide #697: Installation Manager

harrisonhan123, 2019-04-17:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.33.1
PIO IDE: v1.7.0
System: Windows_NT, 6.1.7601, x64

# Exception
```
Error: PIP Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while. 
 C:\Users\Theo\.platformio\penv\Scripts\python.exe: No module named pip

	at f.value.i.runCommand (c:\Users\Theo\.vscode\extensions\platformio.platformio-ide-1.7.0\node_modules\platformio-node-helpers\dist\index.js:1:26217)
	at ChildProcess.h (c:\Users\Theo\.vscode\extensions\platformio.platformio-ide-1.7.0\node_modules\platformio-node-helpers\dist\index.js:1:5916)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (c:\Users\Theo\.vscode\extensions\platformio.platformio-ide-1.7.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

ivankravets, 2019-04-20:
Please remove `C:\Users\Theo\.platformio\penv` folder and restart VSCode.

closed 2019-04-20 by ivankravets

harrisonhan123, 2019-04-27:
Hi sir,

I've tried your suggestion, but still same problem.
any other suggestions?

Thanks & Regards
Harrison