platformio/platformio-vscode-ide #701: Start PIO Home Server

sbehnke, 2019-04-18:
# Description of problem
Was just trying to open a workspace.

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.33.1
PIO IDE: v1.7.0
System: Darwin, 18.5.0, x64

# Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at P.Promise (/Users/sbehnke/.vscode/extensions/platformio.platformio-ide-1.7.0/node_modules/platformio-node-helpers/dist/index.js:1:14992)
	at ChildProcess.h (/Users/sbehnke/.vscode/extensions/platformio.platformio-ide-1.7.0/node_modules/platformio-node-helpers/dist/index.js:1:5916)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

iwaresk, 2019-04-18:
I have the same Problem on Ubuntu 19.04.
On the first start i got an Error that Python 2.7 is not installed because Ubuntu 19.04 uses Python 3. So i installed Python 2.7.16 and got this error.

Error: Error: Detected unknown package 'contrib-pysite'

	at P.Promise (/home/ingo/.vscode/extensions/platformio.platformio-ide-1.7.0/node_modules/platformio-node-helpers/dist/index.js:1:14992)
	at ChildProcess.h (/home/ingo/.vscode/extensions/platformio.platformio-ide-1.7.0/node_modules/platformio-node-helpers/dist/index.js:1:5916)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)

iwaresk, 2019-04-18:
I tried PlatformIO with Atom but i got the same error.

# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.36.0
PIO IDE: v2.3.1
System: Linux, 5.0.0-13-generic, x64

# Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

    at /home/ingo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:15270
    at ChildProcess.h (/home/ingo/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/dist/index.js:1:5904)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at maybeClose (internal/child_process.js:925:16)
    at Socket.stream.socket.on (internal/child_process.js:346:11)
    at emitOne (events.js:116:13)
    at Socket.emit (events.js:211:7)
    at Pipe._handle.close [as _onclose] (net.js:554:12)
```

sushicodeur, 2019-04-18:
Same problem on a fresh install of vscode + platform io

# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.33.1
PIO IDE: v1.7.0
System: Darwin, 18.2.0, x64

# Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at P.Promise (/Users/alaindiart/.vscode/extensions/platformio.platformio-ide-1.7.0/node_modules/platformio-node-helpers/dist/index.js:1:14992)
	at ChildProcess.h (/Users/alaindiart/.vscode/extensions/platformio.platformio-ide-1.7.0/node_modules/platformio-node-helpers/dist/index.js:1:5916)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

wtfuzz, 2019-04-19:
Same problem here

# Configuration

VSCode: 1.33.1
PIO IDE: v1.7.0
System: Darwin, 18.5.0, x64

# Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at P.Promise (/Users/fuzz/.vscode/extensions/platformio.platformio-ide-1.7.0/node_modules/platformio-node-helpers/dist/index.js:1:14992)
	at ChildProcess.h (/Users/fuzz/.vscode/extensions/platformio.platformio-ide-1.7.0/node_modules/platformio-node-helpers/dist/index.js:1:5916)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

stephaun, 2019-04-19:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.33.1
PIO IDE: v1.7.0
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at P.Promise (C:\Users\sjg\.vscode\extensions\platformio.platformio-ide-1.7.0\node_modules\platformio-node-helpers\dist\index.js:1:14992)
	at ChildProcess.h (C:\Users\sjg\.vscode\extensions\platformio.platformio-ide-1.7.0\node_modules\platformio-node-helpers\dist\index.js:1:5916)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (C:\Users\sjg\.vscode\extensions\platformio.platformio-ide-1.7.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

ivankravets, 2019-04-19:
Duplicate of https://github.com/platformio/platformio-core/issues/2341

closed 2019-04-19 by ivankravets