platformio/platformio-vscode-ide #722: Start PIO Home Server

jethro2533, 2019-04-19:
# Description of problem
I have tried installing to two different Windows machines. One with Python installed and one without. Same error and PlatformIO just continues to try and load, without actually completing the task.

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.33.1
PIO IDE: v1.7.0
System: Windows_NT, 10.0.17763, x64

# Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at P.Promise (C:\Users\Jeff\.vscode\extensions\platformio.platformio-ide-1.7.0\node_modules\platformio-node-helpers\dist\index.js:1:14992)
	at ChildProcess.h (C:\Users\Jeff\.vscode\extensions\platformio.platformio-ide-1.7.0\node_modules\platformio-node-helpers\dist\index.js:1:5916)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (C:\Users\Jeff\.vscode\extensions\platformio.platformio-ide-1.7.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

ivankravets, 2019-04-19:
Duplicate of https://github.com/platformio/platformio-core/issues/2341

closed 2019-04-19 by ivankravets

jethro2533, 2019-04-19:
Just had success installing PlatformIO. I removed VSC ver 1.33 (the latest) and installed ver 1.26.1 . Installed the PlatformIO -_IDE and then closed up VSC. Openned it up again and VSC upgraded to 1.33 with no issues as yet with PlatformIO.

ivankravets, 2019-04-19:
Please upgrade VSCode back to the latest version. This was our issue, not VSCode.