platformio/platformio-vscode-ide #728: Start PIO Home Server

Maxzprint, 2019-04-19:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.33.1
PIO IDE: v1.7.0
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at P.Promise (C:\Users\Maxzprint\.vscode\extensions\platformio.platformio-ide-1.7.0\node_modules\platformio-node-helpers\dist\index.js:1:14992)
	at ChildProcess.h (C:\Users\Maxzprint\.vscode\extensions\platformio.platformio-ide-1.7.0\node_modules\platformio-node-helpers\dist\index.js:1:5916)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (C:\Users\Maxzprint\.vscode\extensions\platformio.platformio-ide-1.7.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

Maxzprint, 2019-04-19:
Thought this was an atom issue but now it has also occured in Visual Studio

ivankravets, 2019-04-19:
Duplicate of https://github.com/platformio/platformio-core/issues/2341

closed 2019-04-19 by ivankravets