platformio/platformio-vscode-ide #732: Issue loading PlatformIO

PabloAbraham, 2019-04-19:
Hi guys, I have an issue loading PlatformIO in my laptop. It was working fine before but it seems that some of the last updates are generating some issues.

[image]

The issue reported on the screen is:
exceptions.AttributeError: 'int' object has no attribute 'splitlines'

Here you can see the versions that I am using:
[image]

PlatformIO Version: 1.7.0
Python Version: 2019.3.6558

it seems that the issue is with this line:
C:\Users\Admin\.platformio\packages\contrib-pysite\twisted\web\http_headers.py:40 in _sanitizeLinearWhitespace

I didn't make any modifications on configurations or .py files. Just a typical install (and it was working fine before.)

Hokutosei, 2019-04-20:
same problem...
[image]

ivankravets, 2019-04-20:
Please run PlatformIO >Update All or Open PlatformIO IDE and  run `pio update`. Restart VSCode

closed 2019-04-20 by ivankravets

PabloAbraham, 2019-04-20:
Thanks Ivan! Your recommendation worked fine!

I updated as you suggested:
[image]

and now it is back:

[image]

Thanks for your quick response!

Regards

Pablo

Hokutosei, 2019-04-22:
@ivankravets thanks, it is now fixed.