platformio/platformio-vscode-ide #733: Installation Manager

Emilleopold, 2019-04-19:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.33.1
PIO IDE: v1.7.0
System: Darwin, 18.5.0, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "/Users/mschuessler/.platformio/.cache/tmp-2947GjsHb02XQ8KY/virtualenv-16.1.0/src/virtualenv.py", line 33, in <module>
    import zlib
ImportError: No module named zlib

	at t.value (/Users/mschuessler/.vscode/extensions/platformio.platformio-ide-1.7.0/node_modules/platformio-node-helpers/dist/index.js:1:25327)
```

ivankravets, 2019-04-20:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

Emilleopold, 2019-04-20:
Dear Ivan,

I have updated Python and Python3 with Homebrew and installed virtualenv with PIP. After that I was able to install Platformio.
But if I want to compile the Arduino blink example I get the message that „Arduino.h“ is not found. There seems to be a problem with paths.
I will deinstall and again install the platformio.
Thanks.

Mit freundlichen Grüßen / Best regards
 
Manfred Schüssler
 
MESTRON MESS- UND REGELTECHNIK GMBH
Drosselweg 17
72666 Neckartailfingen
E-Mail : ms@mestron.de 
E-Mail : manfredschuessler@me.com
Telefon / Fon : +49(0)7127 / 9552 95-0
Telefax/ Fax : +49(0)7127 / 9552 95-9
Handy / Mobil : +49(0)160/90364611
 
Geschäftsführer / Managing director : Manfred Schüssler
Sitz der Gesellschaft / Registered office : Neckartailfingen
Registriergericht / Company registration  : Stuttgart HRB 300709
WEEE-Reg.-Nr. : DE 144994535

ivankravets, 2019-04-20:
Could you provide your `platformio.ini`?

closed 2019-05-28 by ivankravets