platformio/platformio-vscode-ide #735: PlatformIO: won't compile minimal project setup due to multiple OSError exceptions

HouzuoGuo, 2019-04-20:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**:
Windows 10 1803

**PlatformIO Version** (`platformio --version`):
3.6.6

### Description of problem
PlatformIO won't compile a bare minimal project created for TTGO T-Beam board.

#### Steps to Reproduce

1. Use PlatformIO VSCode plugin to create a bare minimal project.
2. In command prompt, Navigate into the project's top level directory and run `platformio.exe run`.
3. Observe multiple exceptions that eventually lead to task failure.

### Actual Results

<pre>
C:\Users\Houzuo Guo\Documents\PlatformIO\Projects\tbeam-pio>"C:\Users\Houzuo Guo\.platformio\penv\Scripts\platformio.exe" run
Processing ttgo-t-beam (platform: espressif32; board: ttgo-t-beam; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-t-beam.html
PLATFORM: Espressif 32 > TTGO T-Beam
HARDWARE: ESP32 240MHz 1.25MB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 26 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs\ttgo-t-beam\src\main.cpp.o
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe': No child processes
Generating partitions .pioenvs\ttgo-t-beam\partitions.bin
Archiving .pioenvs\ttgo-t-beam\libFrameworkArduinoVariant.a
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe /C ""c:\users\houzuo guo\.platformio\penv\scripts\python.exe" "C:\Users\Houzuo Guo\.platformio\packages\framework-arduinoespressif32\tools\gen_esp32part.py" -q "C:\Users\Houzuo Guo\.platformio\packages\framework-arduinoespressif32\tools\partitions\default.csv" ".pioenvs\ttgo-t-beam\partitions.bin""': No child processes
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe /C "xtensa-esp32-elf-ar rc ".pioenvs\ttgo-t-beam\libFrameworkArduinoVariant.a""': No child processes
Compiling .pioenvs\ttgo-t-beam\FrameworkArduino\Esp.cpp.o
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe': No child processes
Compiling .pioenvs\ttgo-t-beam\FrameworkArduino\FunctionalInterrupt.cpp.o
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe': No child processes
Compiling .pioenvs\ttgo-t-beam\FrameworkArduino\HardwareSerial.cpp.o
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe': No child processes
Compiling .pioenvs\ttgo-t-beam\FrameworkArduino\IPAddress.cpp.o
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe': No child processes
*** [.pioenvs\ttgo-t-beam\src\main.cpp.o] Error 127
*** [.pioenvs\ttgo-t-beam\partitions.bin] Error 127Compiling .pioenvs\ttgo-t-beam\FrameworkArduino\IPv6Address.cpp.o
*
** [.pioenvs\ttgo-t-beam\libFrameworkArduinoVariant.a] Error 127
*** [.pioenvs\ttgo-t-beam\FrameworkArduino\Esp.cpp.o] Error 127
*** [.pioenvs\ttgo-t-beam\FrameworkArduino\FunctionalInterrupt.cpp.o] Error 127
*** [.pioenvs\ttgo-t-beam\FrameworkArduino\HardwareSerial.cpp.o] Error 127
*** [.pioenvs\ttgo-t-beam\FrameworkArduino\IPAddress.cpp.o] Error 127
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe': No child processes
*** [.pioenvs\ttgo-t-beam\FrameworkArduino\IPv6Address.cpp.o] Error 127
============================================================================ [ERROR] Took 1.82 seconds ============================================================================

</pre>

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:ttgo-t-beam]
platform = espressif32
board = ttgo-t-beam
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

void setup() {
}

void loop() {

}
```

ivankravets, 2019-04-20:
Please try the next:

1) Open `cmd.exe` and navigate to `%USER_HOME%/.platformio/penv`
2) Run `Scripts\activate`
3) Run `pio run -d %path/to/platformio/project/direcotry%

Does it work?

HouzuoGuo, 2019-04-21:
Thanks @ivankravets. Here is the symptom:

<pre>
C:\Users\Houzuo Guo>cd .platformio

C:\Users\Houzuo Guo\.platformio>dir
 Volume in drive C has no label.
 Volume Serial Number is 50B3-A68A

 Directory of C:\Users\Houzuo Guo\.platformio

21/04/2019  16.38    <DIR>          .
21/04/2019  16.38    <DIR>          ..
21/04/2019  16.37    <DIR>          .cache
21/04/2019  16.37               215 appstate.json
21/04/2019  16.38               442 homestate.json
20/04/2019  12.08    <DIR>          lib
21/04/2019  16.38    <DIR>          packages
20/04/2019  12.04    <DIR>          penv
21/04/2019  16.36    <DIR>          platforms
20/04/2019  12.04    <DIR>          python27
               2 File(s)            657 bytes
               8 Dir(s)  249 107 955 712 bytes free

C:\Users\Houzuo Guo\.platformio>cd penv

C:\Users\Houzuo Guo\.platformio\penv>Scripts\activate

(penv) C:\Users\Houzuo Guo\.platformio\penv>pio
Usage: pio [OPTIONS] COMMAND [ARGS]...

Options:
  --version          Show the version and exit.
  -f, --force        Force to accept any confirmation prompts.
  -c, --caller TEXT  Caller ID (service).
  -h, --help         Show this message and exit.

Commands:
  account   Manage PIO Account
  boards    Embedded Board Explorer
  ci        Continuous Integration
  debug     PIO Unified Debugger
  device    Monitor device or list existing
  home      PIO Home
  init      Initialize PlatformIO project or update existing
  lib       Library Manager
  platform  Platform Manager
  remote    PIO Remote
  run       Process project environments
  settings  Manage PlatformIO settings
  test      Local Unit Testing
  update    Update installed platforms, packages and libraries
  upgrade   Upgrade PlatformIO to the latest version

(penv) C:\Users\Houzuo Guo\.platformio\penv>pio run -d  ..\..\Documents\PlatformIO\Projects\tbeam-pio
Processing ttgo-t-beam (platform: espressif32; board: ttgo-t-beam; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-t-beam.html
PLATFORM: Espressif 32 > TTGO T-Beam
HARDWARE: ESP32 240MHz 1.25MB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 26 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs\ttgo-t-beam\src\main.cpp.o
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe': No child processes
Generating partitions .pioenvs\ttgo-t-beam\partitions.bin
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe /C ""c:\users\houzuo guo\.platformio\penv\scripts\python.exe" "C:\Users\Houzuo Guo\.platformio\packages\framework-arduinoespressif32\tools\gen_esp32part.py" -q "C:\Users\Houzuo Guo\.platformio\packages\framework-arduinoespressif32\tools\partitions\default.csv" ".pioenvs\ttgo-t-beam\partitions.bin""': No child processes
Archiving .pioenvs\ttgo-t-beam\libFrameworkArduinoVariant.a
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe /C "xtensa-esp32-elf-ar rc ".pioenvs\ttgo-t-beam\libFrameworkArduinoVariant.a""': No child processes
Compiling .pioenvs\ttgo-t-beam\FrameworkArduino\Esp.cpp.o
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe': No child processes
Compiling .pioenvs\ttgo-t-beam\FrameworkArduino\FunctionalInterrupt.cpp.o
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe': No child processes
Compiling .pioenvs\ttgo-t-beam\FrameworkArduino\HardwareSerial.cpp.o
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe': No child processes
Compiling .pioenvs\ttgo-t-beam\FrameworkArduino\IPAddress.cpp.o
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe': No child processes
*** [.pioenvs\ttgo-t-beam\src\main.cpp.o] Error 127
C*ompiling .pioenvs\ttgo-t-beam\FrameworkArduino\IPv6Address.cpp.o** [.pioenvs\ttgo-t-beam\partitions.bin] Error 127

*** [.pioenvs\ttgo-t-beam\libFrameworkArduinoVariant.a] Error 127
*** [.pioenvs\ttgo-t-beam\FrameworkArduino\Esp.cpp.o] Error 127
*** [.pioenvs\ttgo-t-beam\FrameworkArduino\FunctionalInterrupt.cpp.o] Error 127
*** [.pioenvs\ttgo-t-beam\FrameworkArduino\HardwareSerial.cpp.o] Error 127
*** [.pioenvs\ttgo-t-beam\FrameworkArduino\IPAddress.cpp.o] Error 127
unknown OSError exception code 10 - 'C:\Windows\System32\cmd.exe': No child processes
*** [.pioenvs\ttgo-t-beam\FrameworkArduino\IPv6Address.cpp.o] Error 127
============================================== [ERROR] Took 2.04 seconds ==============================================

(penv) C:\Users\Houzuo Guo\.platformio\penv>
</pre>

ivankravets, 2019-05-28:
Did you resolve this issue?

HouzuoGuo, 2019-05-29:
Unfortunately there's not been any progress at all, I am using VSCode Arduino plugin as a fall back atm.

golfromeo-fr, 2019-08-16:
I had a very similar issue, probably a python version problem

remove any version of python installed (that can be accessed from PATH)
remove .plateformio and .vscode
restart vscode

ivankravets, 2019-08-16:
Did it help you?

HouzuoGuo, 2019-08-17:
Not really, there isn't another copy of python installed on this system.

closed 2020-09-12 by HouzuoGuo