platformio/platformio-vscode-ide #736: when I click the home button, it    /Users/qinshaobo/.platformio/packages/contrib-pysite/twisted/web/server.py:199 in process  198 self. encoder = encoder 199 self.render (resrc) 200 except:  /Users/qinshaobo/.platformio/packages/contrib-pysite/twisted/web/server.py:259 in render  258 try: 259 body = resrc.render (self) 260 except UnsupportedMethod as e:  /Users/qinshaobo/.platformio/packages/contrib-pysite/twisted/web/resource.py:250 in render  249 raise UnsupportedMethod (allowedMethods) 250 return m(request)  251

joyspce, 2019-04-21:
/Users/qinshaobo/.platformio/packages/contrib-pysite/twisted/web/server.py:199 in
process

198 self. encoder = encoder
199 self.render (resrc)
200 except:

/Users/qinshaobo/.platformio/packages/contrib-pysite/twisted/web/server.py:259 in
render

258 try:
259 body = resrc.render (self)
260 except UnsupportedMethod as e:

/Users/qinshaobo/.platformio/packages/contrib-pysite/twisted/web/resource.py:250 in
render

249 raise UnsupportedMethod (allowedMethods)
250 return m(request)

251

joyspce, 2019-04-21:
I delete the ~/.platformio directory, and reinstall code and platformio, it works.

AlyoshaVasilieva, 2019-04-21:
Same issue. Win10, VSCode 1.33.1. Updated VSCode to 1.33.1 after it happened, was originally on 1.33. Happened immediately after updating Adafruit MQTT Library to 1.0.0 and Espressif32 to 1.8.0 and restarting VSCode.

```
exceptions.AttributeError: 'int' object has no attribute 'splitlines'
C:\Users\L\.platformio\packages\contrib-pysite\twisted\web\server.py:199 in process
198                    self._encoder = encoder
199            self.render(resrc)
200        except:
C:\Users\L\.platformio\packages\contrib-pysite\twisted\web\server.py:259 in render
258        try:
259            body = resrc.render(self)
260        except UnsupportedMethod as e:
C:\Users\L\.platformio\packages\contrib-pysite\twisted\web\resource.py:250 in render
249            raise UnsupportedMethod(allowedMethods)
250        return m(request)
251
<string>:448 in render_GET
447
448
449
C:\Users\L\.platformio\packages\contrib-pysite\twisted\web\http.py:1271 in setHeader
1270        """
1271        self.responseHeaders.setRawHeaders(name, [value])
1272
C:\Users\L\.platformio\packages\contrib-pysite\twisted\web\http_headers.py:220 in setRawHeaders
219        encodedValues = [_sanitizeLinearWhitespace(v)
220                         for v in self._encodeValues(values)]
221
C:\Users\L\.platformio\packages\contrib-pysite\twisted\web\http_headers.py:40 in _sanitizeLinearWhitespace
39    """
40    return b' '.join(headerComponent.splitlines())
41
exceptions.AttributeError: 'int' object has no attribute 'splitlines'
```

mbrauerImmonet, 2019-04-21:
deleting .platformio works on windows too.

ivankravets, 2019-04-22:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/732
P.S: Just open PlatformIO IDE terminal and type `pio update`. Restart VSCode.

closed 2019-04-22 by ivankravets