platformio/platformio-vscode-ide #737: Installation Manager

MarkMan0, 2019-04-21:
# Description of problem

2-3 days ago it worked without a problem. Today I wanted to download a library, and it throwed some exceptions:
exceptions.AttributeError: 'int' object has no attribute 'splitlines'
And a bunch of other stuff I can't remember
Couldn't even go to the home screen.
So I've decided to reinstall the extension, and now I'm here...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.33.1
PIO IDE: v1.7.1
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: If you use Antivirus, it can block PlatformIO Installer. Try to disable it for a while.

Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\Mark\.platformio\.cache\tmp-11824M6I0R5GeCfQ8\virtualenv-16.1.0\src\virtualenv.py", line 2462, in <module>
    main()
  File "C:\Users\Mark\.platformio\.cache\tmp-11824M6I0R5GeCfQ8\virtualenv-16.1.0\src\virtualenv.py", line 762, in main
    symlink=options.symlink,
  File "C:\Users\Mark\.platformio\.cache\tmp-11824M6I0R5GeCfQ8\virtualenv-16.1.0\src\virtualenv.py", line 1015, in create_environment
    install_wheel(to_install, py_executable, search_dirs, download=download)
  File "C:\Users\Mark\.platformio\.cache\tmp-11824M6I0R5GeCfQ8\virtualenv-16.1.0\src\virtualenv.py", line 968, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "C:\Users\Mark\.platformio\.cache\tmp-11824M6I0R5GeCfQ8\virtualenv-16.1.0\src\virtualenv.py", line 812, in call_subprocess
    env=env,
  File "C:\Users\Mark\.platformio\python27\Lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "C:\Users\Mark\.platformio\python27\Lib\subprocess.py", line 640, in _execute_child
    startupinfo)
WindowsError: [Error 5] Access is denied

Please use the *system* python to run this script
Installing setuptools, pip, wheel...

  Error [Error 5] Access is denied while executing command C:\Users\Mark\.platformio\penv\Scripts - setuptools pip wheel
...Installing setuptools, pip, wheel...done.

	at t.value (C:\Users\Mark\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:25605)
```

MarkMan0, 2019-04-21:
I've reinstalled and restarted several times, and I have PlatformIO installed now.
But it's still throwing the same errors as before:

`    web.Server Traceback (most recent call last):
    exceptions.AttributeError: 'int' object has no attribute 'splitlines'
    C:\Users\Mark\.platformio\packages\contrib-pysite\twisted\web\server.py:199 in process
    198                    self._encoder = encoder
    199            self.render(resrc)
    200        except:
    C:\Users\Mark\.platformio\packages\contrib-pysite\twisted\web\server.py:259 in render
    258        try:
    259            body = resrc.render(self)
    260        except UnsupportedMethod as e:
    C:\Users\Mark\.platformio\packages\contrib-pysite\twisted\web\resource.py:250 in render
    249            raise UnsupportedMethod(allowedMethods)
    250        return m(request)
    251
    <string>:448 in render_GET
    447
    448
    449
    C:\Users\Mark\.platformio\packages\contrib-pysite\twisted\web\http.py:1271 in setHeader
    1270        """
    1271        self.responseHeaders.setRawHeaders(name, [value])
    1272
    C:\Users\Mark\.platformio\packages\contrib-pysite\twisted\web\http_headers.py:220 in setRawHeaders
    219        encodedValues = [_sanitizeLinearWhitespace(v)
    220                         for v in self._encodeValues(values)]
    221
    C:\Users\Mark\.platformio\packages\contrib-pysite\twisted\web\http_headers.py:40 in _sanitizeLinearWhitespace
    39    """
    40    return b' '.join(headerComponent.splitlines())
    41
    exceptions.AttributeError: 'int' object has no attribute 'splitlines'`

Naarlack, 2019-04-22:
I had this exact same issue. Although I think I have made things worse by trying to fix it...

Hope you get a response!

ivankravets, 2019-04-22:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/732
P.S: Just open PlatformIO IDE terminal and type `pio update`. Restart VSCode.

closed 2019-04-22 by ivankravets