platformio/platformio-vscode-ide #738: pio home

BogdanSav, 2019-04-21:
web.Server Traceback (most recent call last):
exceptions.AttributeError: 'int' object has no attribute 'splitlines'

BogdanSav, 2019-04-21:
C:\Users\8560\.platformio\packages\contrib-pysite\twisted\web\server.py:199 in process
198                    self._encoder = encoder
199            self.render(resrc)
200        except:
C:\Users\8560\.platformio\packages\contrib-pysite\twisted\web\server.py:259 in render
258        try:
259            body = resrc.render(self)
260        except UnsupportedMethod as e:
C:\Users\8560\.platformio\packages\contrib-pysite\twisted\web\resource.py:250 in render
249            raise UnsupportedMethod(allowedMethods)
250        return m(request)
251
<string>:448 in render_GET
447
448
449
C:\Users\8560\.platformio\packages\contrib-pysite\twisted\web\http.py:1271 in setHeader
1270        """
1271        self.responseHeaders.setRawHeaders(name, [value])
1272
C:\Users\8560\.platformio\packages\contrib-pysite\twisted\web\http_headers.py:220 in setRawHeaders
219        encodedValues = [_sanitizeLinearWhitespace(v)
220                         for v in self._encodeValues(values)]
221
C:\Users\8560\.platformio\packages\contrib-pysite\twisted\web\http_headers.py:40 in _sanitizeLinearWhitespace
39    """
40    return b' '.join(headerComponent.splitlines())
41

oliver-davies, 2019-04-21:
Yep having the same issue! On three machines, nothing seems to resolve it. Please fix your stuff platformio.

CornyjK, 2019-04-22:
After latest update opening PIO home stucks at `Loading...`.

ivankravets, 2019-04-22:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/732
P.S: Just open PlatformIO IDE terminal and type `pio update`. Restart VSCode.

closed 2019-04-22 by ivankravets