platformio/platformio-vscode-ide #739: Installation Manager

Naarlack, 2019-04-22:
# Description of problem
Have been using PIO for a few days with no issues. Today I tried to open PIO Home but it appears with all sorts of errors, something about ".splitlines"? I restarted my laptop, no effect. I uninstalled and reinstalled PIO, no effect. I uninstalled PIO and deleted the .platformio folder then tried to reinstall, and now I am here...

Any help would be appreciated

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.33.1
PIO IDE: v1.7.1
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Conda Virtualenv: 
CondaHTTPError: HTTP 000 CONNECTION FAILED for url <https://repo.anaconda.com/pkgs/r/noarch/repodata.json.bz2>
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.

If your current network has https://www.anaconda.com blocked, please file
a support request with your network engineering team.

SSLError(MaxRetryError('HTTPSConnectionPool(host=\'repo.anaconda.com\', port=443): Max retries exceeded with url: /pkgs/r/noarch/repodata.json.bz2 (Caused by SSLError("Can\'t connect to HTTPS URL because the SSL module is not available."))'))

	at f.value.i.runCommand (c:\Users\jrrma\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:23662)
	at ChildProcess.h (c:\Users\jrrma\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:5904)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (c:\Users\jrrma\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

ivankravets, 2019-04-22:
Something is going wrong with your Conda installation. Please fix it.

closed 2019-04-22 by ivankravets

Naarlack, 2019-04-22:
Thanks Ivan - you're a legend!