platformio/platformio-vscode-ide #744: Weird base64 code decrypting and executing file

s256, 2019-04-24:
Hi all,
I just found a weird process on my computer:

`platformio/penv/bin/python2.7 -c import base64; exec(base64.b64decode('CiMgQ29weXJpZ2h...MORE BASE64....y5jb20+CiAgICA=')) -f -c vscode home --port 8010 --no-open
`
(https://pastebin.com/7e0Ww96Q)

So I decrypted this base64 string and got this: https://pastebin.com/kWeuttpZ (markup breaks this, sorry)
The File
` fp = open("/tmp/711881635")`
is encrypted with a key
`exec(decrypt_message(os.environ.get("PP8498210RND"), fp.read()));`

On my system (Fedora 29) the OS Variable is empty... lucky me?

File details:
`file /tmp/711881635
/tmp/711881635: ASCII text, with very long lines, with no line terminators`

I asked my colleague to try this on his machine (VM, fresh Ubuntu 18.10) and we found exactly the same. So I made sure it's not my system which is the root of the weird script.

As soon as we uninstalled the platformio extension from vscode, the base64-code process is gone.
That's why I'm posting it here, let me know if this is not the right place.

As this looks super suspicious to me, maybe someone can help shed some light.

So my question is: What is happening here?

I couldn't find anything in the repo or in the issues or on google on the base64 code.

Looking forward to get some insight.

Best
se

ivankravets, 2019-05-07:
This is a known issue. Currently, PIO Plus uses PlatformIO Core Python virtual environment to run a Web server for PIO Home. This issue will be fixed in upcoming PlatformIO Core 4.0. Some part of work is already done.

s256, 2019-05-08:
So, what is hidden in that encrypted file in /tmp/? I mean it just looks like malware...

ivankravets, 2019-05-08:
This is not malware. It is an encrypted proprietary code of PIO Plus tools. As I mentioned above, these tools were open sourced and are planned to be merged into PlatformIO Core 4.0 soon. See the encrypted code here https://github.com/platformio/platformio-core/tree/feature/pio-plus-oss

chris-ault, 2019-05-12:
So "encrypted" is put very lightly here..  Base64 encoding a vigenere-cipher with a system variable key.. only encrypts this from the very lazy.   I caught the system variable that's local to the process with Process Hacker and converted your execute script with a print showing another Python script that base64 encodes some other python modules. 
load_pysite
config
exception
requests_threads
cmd_account
cmd_test
project_sync
remote_client
remote_agent
cmd_remote
cmd_debug'
cmd_home
main

This whole encryption process was probably done with https://github.com/liftoff/pyminifier pretty sweet little tool, will look into it some more!

ivankravets, 2019-05-28:
Fixed in the latest PIO Core which has open sourced PIO Home. Please upgrade http://docs.platformio.org/en/latest/installation.html#development-version

closed 2019-05-28 by ivankravets

MikeWent, 2019-06-04:
> `exec(decrypt_message(os.environ.get("PP8498210RND"), fp.read()));`
>
> On my system (Fedora 29) the OS Variable is empty... lucky me?

I get `PP8498210RND=0HZMW1LL5N02M` when I do `cat /proc/PID/environ`.

> I mean it just looks like malware...

I've also found this behavior very suspicious.

@ivankravets, is there any config value or a switch that turns off this feature? I don't use PlatformIO Plus and have no plans on using it in the future yet.

ivankravets, 2019-06-04:
@MikeWent please switch to development version of PlatformIO Core 4.0, see instruction here http://docs.platformio.org/en/latest/ide/vscode.html#platformio-ide-usedevelopmentpiocore

I also sent an email.