platformio/platformio-vscode-ide #749: PlatformioIO-IDE Installation Issue: .platformio/penv/bin/python22.7: No such file or directory

jakubczaplicki, 2019-04-26:
# Description of problem

The installation tries to use, what seems to be a wrong file name: `python22.7`, instead of `python2.7`

# Configuration

VSCode: 1.33.1
PIO IDE: v1.7.1
System: Darwin, 18.5.0, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: python2: posix_spawn: /Users/jakubczaplicki/.platformio/penv/bin/python22.7: No such file or directory

New python executable in /Users/jakubczaplicki/.platformio/penv/bin/python2
Also creating executable in /Users/jakubczaplicki/.platformio/penv/bin/python
ERROR: The executable /Users/jakubczaplicki/.platformio/penv/bin/python2 is not functioning
ERROR: It thinks sys.prefix is u'/' (should be u'/Users/jakubczaplicki/.platformio/penv')
ERROR: virtualenv is not compatible with this system or executable

	at t.value (/Users/jakubczaplicki/.vscode/extensions/platformio.platformio-ide-1.7.1/node_modules/platformio-node-helpers/dist/index.js:1:25605)
```

ivankravets, 2019-07-13:
Please reopen if you still need help

closed 2019-07-13 by ivankravets

mhyousefi, 2019-10-12:
I have this issue when creating a Python2 `virtualenv` for a project. I have oppenned an issue [here](https://github.com/pypa/virtualenv/issues/1427). Can you help?

ivankravets, 2019-10-12:
Just install Python 3 and it has built-in `python3 -m venv`. PlatformIO will handle it automatically.