platformio/platformio-vscode-ide #760: Error: The PlatformIO task detection didn't contribute a task

mklengel, 2019-05-02:
Error on my system (macOS 10.14.4, VS Code 1.33.1, PlatformIO IDE 1.7.1) during the Build:

```
Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "task": "Upload",
    "problemMatcher": [
        "$platformio"
    ]
}
The task will be ignored.
```

Contents of platformio.ini:
```
[env:uno]
platform = atmelavr
board = uno
framework = Arduino
```

Build:
```
Executing task in folder mkBlinkyMega2560: platformio run <

Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz 8KB RAM (248KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 5 compatible libraries
Scanning dependencies...
No dependencies
Checking size .pioenvs/megaatmega2560/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   0.1% (used 9 bytes from 8192 bytes)
PROGRAM: [          ]   0.6% (used 1568 bytes from 253952 bytes)
======================================================================================== [SUCCESS] Took 0.76 seconds ========================================================================================

Terminal will be reused by tasks, press any key to close it
```

Testing the Upload task in the VS Code terminal works (no board attached):

```
pio run --target upload
Processing uno (platform: atmelavr; board: uno; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/uno.html
PLATFORM: Atmel AVR > Arduino Uno
HARDWARE: ATMEGA328P 16MHz 2KB RAM (31.50KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 5 compatible libraries
Scanning dependencies...
No dependencies
Checking size .pioenvs/uno/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   0.4% (used 9 bytes from 2048 bytes)
PROGRAM: [          ]   2.9% (used 930 bytes from 32256 bytes)
Configuring upload protocol...
AVAILABLE: arduino
CURRENT: upload_protocol = arduino
Looking for upload port...
Auto-detected: /dev/cu.SLAB_USBtoUART1
Uploading .pioenvs/uno/firmware.hex
^C*** [upload] Build interrupted.
```

mklengel, 2019-05-06:
Deleting ./mkBlinkyMega2560/.vscode/tasks.json solves the problem. I just don't understand why and when this file was generated and why it caused a problem.

andig, 2019-05-28:
Same problem here on VSCode 1.34 and PIO IDE 2.1.0 (Core 3.6.7). Unfortunately I don't have a `.vscode/tasks.json` to delete.

ivankravets, 2019-05-30:
See http://docs.platformio.org/en/latest/ide/vscode.html#task-runner

closed 2019-05-30 by ivankravets