platformio/platformio-vscode-ide #763: Installation Manager

GeorgeIoak, 2019-05-04:
# Description of problem
Reported in forum that PIO Home was not loading on a Win10 x64 system. It was recommended that the only course of action was to manually delete the %USERPROFILE%\.platformio directory and restart VSCode again.

When I did that, this error popped up when it was trying to reinstall the core

# Configuration

VSCode: 1.33.1
PIO IDE: v1.7.1
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\george\.platformio\.cache\tmp-58401x4dEAG126h1\virtualenv-16.1.0\src\virtualenv.py", line 2462, in <module>
    main()
  File "C:\Users\george\.platformio\.cache\tmp-58401x4dEAG126h1\virtualenv-16.1.0\src\virtualenv.py", line 762, in main
    symlink=options.symlink,
  File "C:\Users\george\.platformio\.cache\tmp-58401x4dEAG126h1\virtualenv-16.1.0\src\virtualenv.py", line 998, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\george\.platformio\.cache\tmp-58401x4dEAG126h1\virtualenv-16.1.0\src\virtualenv.py", line 1317, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\george\.platformio\python27\lib\shutil.py", line 83, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: 'C:\\Users\\george\\.platformio\\penv\\Scripts\\python.exe'

New python executable in C:\Users\george\.platformio\penv\Scripts\python.exe

	at t.value (C:\Users\george\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:25605)
```

ivankravets, 2019-05-06:
> IOError: [Errno 13] Permission denied

There is no permission to create Python virtual environment. Please create it manually -> http://bit.ly/pio-core-virtualenv

ivankravets, 2019-10-28:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1213

closed 2019-10-28 by ivankravets