platformio/platformio-vscode-ide #770: Start PIO Home Server

littlepure2333, 2019-05-09:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.33.1
PIO IDE: v1.7.1
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 37, in <module>
Exception: PIO Plus: Runtime error

	at P.Promise (C:\Users\小纯洁\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:15270)
	at ChildProcess.h (C:\Users\小纯洁\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:5904)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (C:\Users\小纯洁\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

ivankravets, 2019-05-28:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/425

ivankravets, 2019-05-28:
Does it work now?

closed 2019-05-28 by ivankravets