platformio/platformio-vscode-ide #774: Installation Manager

ramihsn, 2019-05-10:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.33.1
PIO IDE: v1.6.1
System: Linux, 4.15.0-48-generic, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "/home/rhassan/.platformio/.cache/tmp-28399AV4S61xyKdTg/virtualenv-16.1.0/src/virtualenv.py", line 2462, in <module>
    main()
  File "/home/rhassan/.platformio/.cache/tmp-28399AV4S61xyKdTg/virtualenv-16.1.0/src/virtualenv.py", line 762, in main
    symlink=options.symlink,
  File "/home/rhassan/.platformio/.cache/tmp-28399AV4S61xyKdTg/virtualenv-16.1.0/src/virtualenv.py", line 998, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "/home/rhassan/.platformio/.cache/tmp-28399AV4S61xyKdTg/virtualenv-16.1.0/src/virtualenv.py", line 1317, in install_python
    shutil.copyfile(executable, py_executable)
  File "/usr/lib/python2.7/shutil.py", line 97, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 26] Text file busy: '/home/rhassan/.platformio/penv/bin/python2.7'

New python executable in /home/rhassan/.platformio/penv/bin/python2.7

	at t.value (/home/rhassan/.vscode/extensions/anucecsit.comp2300-platformio-ide-1.6.1/node_modules/platformio-node-helpers/dist/index.js:1:25327)
```

ivankravets, 2019-05-28:
Please try to remove `/home/rhassan/.platformio/penv` folder and restart IDE.

Does it work now?

ramihsn, 2019-05-28:
@ivankravets yes, Thanks.

closed 2019-05-28 by ivankravets