platformio/platformio-vscode-ide #775: Installation Manager

Vogdanis, 2019-05-11:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.33.1
PIO IDE: v1.7.1
System: Darwin, 16.7.0, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "/Users/Yiannis/.platformio/.cache/tmp-6977FQ0JkYudkdSw/virtualenv-16.1.0/src/virtualenv.py", line 18, in <module>
    import base64
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/base64.py", line 9, in <module>
    import struct
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/struct.py", line 1, in <module>
    from _struct import *
ImportError: No module named _struct

	at t.value (/Users/Yiannis/.vscode/extensions/platformio.platformio-ide-1.7.1/node_modules/platformio-node-helpers/dist/index.js:1:25605)
```

ivankravets, 2019-05-28:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

Vogdanis, 2019-06-06:
`echo $PATH
/usr/local/opt/icu4c/sbin:/usr/local/opt/icu4c/bin:/Users/Yiannis/miniconda3/bin:/Users/Yiannis/.nvm/versions/node/v9.3.0/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/usr/local/sbin:/sbin:/opt/X11/bin:/usr/local/CrossPack-AVR/bin:/Applications/VMware Fusion.app/Contents/Public:/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/
`
echo $PYTHONPATH
/usr/local/lib/python2.7/site-packages:

python --version
Python 2.7.10

pip --version
Traceback (most recent call last):
  File "/usr/local/bin/pip", line 7, in <module>
    from pip._internal import main
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/_internal/__init__.py", line 4, in <module>
    import locale
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/locale.py", line 15, in <module>
    import encodings
ImportError: No module named encodings

pip list
Traceback (most recent call last):
  File "/usr/local/bin/pip", line 7, in <module>
    from pip._internal import main
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/_internal/__init__.py", line 4, in <module>
    import locale
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/locale.py", line 15, in <module>
    import encodings
ImportError: No module named encodings

ivankravets, 2019-06-06:
> /Users/Yiannis/miniconda3/bin

Oh, complicated environment. It seems that you have broken Python installation or `miniconda` broke it.

closed 2019-07-13 by ivankravets