platformio/platformio-vscode-ide #785: ESP32 crashes after `framework-arduinoespressif32` update

GeorgeFlorian, 2019-05-16:
Hello !
I had a piece of code that yesterday worked without any errors, but now it crashes the ESP.
The only thing I did was to update `framework-arduinoespressif32` inside Platformio IDE.

```
//------------------------- fileReadLines()
void fileReadLines(File file, String x[]) {
  int i = 0;
    while(file.available()){
        Serial.println((String)"File size: " + file.size());
        String line= file.readStringUntil('\n');
        line.trim();
        x[i] = line; // This is where it crashes after i
        i++;
        logOutput(line);
    }    
}
```

The problem is that the `file.size()` stays the same, and `while(file.available())` goes on until I reach the limit of `String x[]`.

How was it that up until the update it worked and now it doesn't ?

Is `String line= file.readStringUntil('\n');` not working like before ?

ivankravets, 2019-05-16:
Please switch to the stable version of dev/platform or report to https://github.com/espressif/arduino-esp32/issues

closed 2019-05-16 by ivankravets

GeorgeFlorian, 2019-05-16:
> Please switch to the stable version of dev/platform or report to https://github.com/espressif/arduino-esp32/issues

I did both of those things. I also thought it may be a Platformio issue. Thank you !