platformio/platformio-vscode-ide #795: Installation%20Manager

DavidLongley, 2019-05-22:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.34.0%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Windows_NT%2C%2010.0.17134%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20Could%20not%20create%20PIO%20Core%20Virtual%20Environment.%20Please%20create%20it%20manually%20-%3E%20http%3A%2F%2Fbit.ly%2Fpio-core-virtualenv%20%0A%20Error%3A%20Virtualenv%20Create%3A%20Error%3A%20spawn%20UNKNOWN%0A%0A%09at%20t.value%20(c%3A%5CUsers%5Cdlongley%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A25605)%0A%09at%20process._tickCallback%20(internal%2Fprocess%2Fnext_tick.js%3A68%3A7)%0A%60%60%60%0A

ivankravets, 2019-05-28:
Did you resolve this issue?

DavidLongley, 2019-05-28:
No, still an issue.

ivankravets, 2019-05-28:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

DavidLongley, 2019-05-28:
C:\Users\dlongley>echo %PATH%
C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program
Files (x86)\Microsoft SQL Server\150\DTS\Binn\;C:\Program Files\Microsoft
VS Code\bin;C:\Program Files\dotnet\;C:\Program Files\Microsoft SQL
Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client
SDK\ODBC\170\Tools\Binn\;c:\Program Files (x86)\Microsoft SQL
Server\90\Tools\binn\;C:\Program Files (x86)\Microsoft SQL
Server\100\Tools\Binn\;C:\Program Files\Microsoft SQL
Server\100\Tools\Binn\;C:\Program Files\Microsoft SQL
Server\100\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL
Server\100\Tools\Binn\VSShell\Common7\IDE\;C:\Program Files (x86)\Microsoft
Visual Studio 9.0\Common7\IDE\PrivateAssemblies\;C:\Program Files
(x86)\Microsoft SQL
Server\100\DTS\Binn\;C:\Users\dlongley\AppData\Local\Microsoft\WindowsApps;

C:\Users\dlongley>echo %PYTHONPATH%
%PYTHONPATH%

C:\Users\dlongley>python --version
'python' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\dlongley>pip --version
'pip' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\dlongley>pip list
'pip' is not recognized as an internal or external command,
operable program or batch file.

ivankravets, 2019-05-28:
Do you have antivirus tools in a system? Maybe they block our installer? Could you reproduce this issue after IDE restart?

closed 2019-07-13 by ivankravets