platformio/platformio-vscode-ide #803: Start%20PIO%20Home%20Server

oxoo2a, 2019-05-28:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.34.0%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Darwin%2C%2018.6.0%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20Traceback%20(most%20recent%20call%20last)%3A%0A%20%20File%20%22%3Cstring%3E%22%2C%20line%201%2C%20in%20%3Cmodule%3E%0A%20%20File%20%22%3Cstring%3E%22%2C%20line%2038%2C%20in%20%3Cmodule%3E%0A%20%20File%20%22%3Cstring%3E%22%2C%20line%2039%2C%20in%20%3Cmodule%3E%0A%20%20File%20%22%3Cstring%3E%22%2C%20line%2018%2C%20in%20load_module%0A%20%20File%20%22%3Cstring%3E%22%2C%20line%2023%2C%20in%20%3Cmodule%3E%0AImportError%3A%20cannot%20import%20name%20get_project_test_dir%0A%0A%09at%20P.Promise%20(%2FUsers%2Fsturm%2F.vscode%2Fextensions%2Fplatformio.platformio-ide-1.7.1%2Fnode_modules%2Fplatformio-node-helpers%2Fdist%2Findex.js%3A1%3A15270)%0A%09at%20ChildProcess.h%20(%2FUsers%2Fsturm%2F.vscode%2Fextensions%2Fplatformio.platformio-ide-1.7.1%2Fnode_modules%2Fplatformio-node-helpers%2Fdist%2Findex.js%3A1%3A5904)%0A%09at%20ChildProcess.emit%20(events.js%3A182%3A13)%0A%09at%20maybeClose%20(internal%2Fchild_process.js%3A961%3A16)%0A%09at%20Process.ChildProcess._handle.onexit%20(internal%2Fchild_process.js%3A248%3A5)%0A%60%60%60%0A

oxoo2a, 2019-05-28:
I am unable to connect the the server :-(

ivankravets, 2019-05-28:
Please open PlatformIO IDE Terminal and type
```
pio upgrade --dev
```

Restart IDE.

4inn, 2019-05-29:
Hello .. in my case I use PlatformIO on Mac and VSCode .. I had to uninstall the extension, then manually delete the folder (in my case) `~/.platformio` and there install everything again, following the steps suggested here ..

ivankravets, 2019-07-13:
Please upgrade to the latest PIO Core 4.0:
```
$ pio upgrade
```

closed 2019-07-13 by ivankravets