platformio/platformio-vscode-ide #808: Start%20PIO%20Home%20Server

LoneGunnerMe, 2019-05-29:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.34.0%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Windows_NT%2C%2010.0.17134%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20Traceback%20(most%20recent%20call%20last)%3A%0D%0A%20%20File%20%22%3Cstring%3E%22%2C%20line%201%2C%20in%20%3Cmodule%3E%0D%0A%20%20File%20%22%3Cstring%3E%22%2C%20line%2037%2C%20in%20%3Cmodule%3E%0D%0AException%3A%20PIO%20Plus%3A%20Runtime%20error%0D%0A%0A%09at%20P.Promise%20(c%3A%5CUsers%5C%E5%AD%A4%E8%83%86%E6%9E%AA%E6%89%8B%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A15270)%0A%09at%20ChildProcess.h%20(c%3A%5CUsers%5C%E5%AD%A4%E8%83%86%E6%9E%AA%E6%89%8B%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A5904)%0A%09at%20ChildProcess.emit%20(events.js%3A182%3A13)%0A%09at%20ChildProcess.cp.emit%20(c%3A%5CUsers%5C%E5%AD%A4%E8%83%86%E6%9E%AA%E6%89%8B%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Ccross-spawn%5Clib%5Cenoent.js%3A34%3A29)%0A%09at%20maybeClose%20(internal%2Fchild_process.js%3A961%3A16)%0A%09at%20Process.ChildProcess._handle.onexit%20(internal%2Fchild_process.js%3A248%3A5)%0A%60%60%60%0A

demosthien, 2019-05-30:
> %23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fis [...]

FYI LoneGunnerMe, in case you didn't know, you can run that text through a URL decoder like [this one](https://meyerweb.com/eric/tools/dencoder/) to clean up all the %20 stuff.

# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.34.0
PIO IDE: v1.7.1
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 37, in <module>
Exception: PIO Plus: Runtime error

	at P.Promise (c:\Users\孤胆枪手\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:15270)
	at ChildProcess.h (c:\Users\孤胆枪手\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:5904)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (c:\Users\孤胆枪手\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

ivankravets, 2019-05-30:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/425

ivankravets, 2019-05-30:
Does it work if you switch in IDE to a development version of PIO Core?

closed 2019-05-30 by ivankravets