platformio/platformio-vscode-ide #809: Little Snitch blocking VSCode due to "modification of the program".

demosthien, 2019-05-30:
Operating System: OSX 10.14.3

Just an FYI for the devs and anyone else who may come upon this issue (I see it seems something similar happened a couple of times in 2018).

I had previously (about 4 months ago) installed VSCode and PlatformIO (PIO) extension but had shortly thereafter uninstalled.

I recently reinstalled VSCode but when adding PIO it would return an error and not complete the installation. Unfortunately the error returned by VSCode, and presented to "Report" (by directly creating an issue here on Github) was in URL code. I've seen a few of these show up recently so that might be a seperate something the devs should also look at.

After playing around with uninstalling and reinstalling VSCode and PIO (including finding the hidden Users/XXXXX/.platformio/ folder and deleting it) it still returned the same error.

In the end, after scouring the issues here, I found a couple of mentions of Little Snitch blocking VSCode. Sure enough, on checking my Little Snitch it was blocking VSCode showing the problem as;

> The identity check detected a modification of the program. Therefore all of its connections were denied as a precaution.

After allowing VSCode once again PIO has installed correctly. However, I suspect that Little Snitch will block VSCode again should it detect any future modifications to the app by PIO?

Below are three of the error reports I got during this process (I [decoded](https://meyerweb.com/eric/tools/dencoder/) the URL code for clarity, they're probably the same (they look it), but I've pasted them all here just in case there's a difference I don't see.

Hopefully this will help anyone encountering this issue going forward.

==========

# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.34.0
PIO IDE: v1.7.1
System: Darwin, 18.2.0, x64

# Exception
```
Error: PIP Core:   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x107f85190>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/platformio/
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x107f852d0>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/platformio/
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x107f85410>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/platformio/
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x107f85550>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/platformio/
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x107f85690>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/platformio/
  Could not find a version that satisfies the requirement platformio (from versions: )
No matching distribution found for platformio

	at f.value.i.runCommand (/Users/wtf4photography/.vscode/extensions/platformio.platformio-ide-1.7.1/node_modules/platformio-node-helpers/dist/index.js:1:26495)
	at ChildProcess.h (/Users/wtf4photography/.vscode/extensions/platformio.platformio-ide-1.7.1/node_modules/platformio-node-helpers/dist/index.js:1:5904)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:961:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:380:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close [as _onclose] (net.js:596:12)
```
=====

```
Error: PIP Core:   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x1015d2190>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/platformio/
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x1015d22d0>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/platformio/
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x1015d2410>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/platformio/
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x1015d2550>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/platformio/
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x1015d2690>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/platformio/
  Could not find a version that satisfies the requirement platformio (from versions: )
No matching distribution found for platformio

	at f.value.i.runCommand (/Users/wtf4photography/.vscode/extensions/platformio.platformio-ide-1.7.0/node_modules/platformio-node-helpers/dist/index.js:1:26217)
	at ChildProcess.h (/Users/wtf4photography/.vscode/extensions/platformio.platformio-ide-1.7.0/node_modules/platformio-node-helpers/dist/index.js:1:5916)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:961:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:380:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close [as _onclose] (net.js:596:12)
```

=====

```
Error: PIP Core:   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x103a93190>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/platformio/
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x103a932d0>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/platformio/
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x103a93410>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/platformio/
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x103a93550>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/platformio/
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x103a93690>: Failed to establish a new connection: [Errno 64] Host is down',)': /simple/platformio/
  Could not find a version that satisfies the requirement platformio (from versions: )
No matching distribution found for platformio

	at f.value.i.runCommand (/Users/wtf4photography/.vscode/extensions/platformio.platformio-ide-1.7.1/node_modules/platformio-node-helpers/dist/index.js:1:26495)
	at ChildProcess.h (/Users/wtf4photography/.vscode/extensions/platformio.platformio-ide-1.7.1/node_modules/platformio-node-helpers/dist/index.js:1:5904)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

demosthien, 2019-05-30:
### Update...

I have just tried to use PIO to create an example project and it is still getting blocked by Little Snitch.

On trying to install Libraries I got the following;

`PIO Core Call Error: "Error: You are not connected to the Internet.\nIf you build a project first time, we need Internet connection to install all dependencies and toolchains."`

I also get no Platforms, Boards, or example Projects showing up... No internet = No PIO.

Little Snitch is blocking;
> Visual Studio Code via Python

Where;
> /Applications/Visual Studio Code.app/Contents/MacOS/Electron via /Users/wtf4photography/.platformio/penv/bin/python

Priority;
> Extra High – The identity check detected a modification of the program. Therefore all of its connections were denied as a precaution.

There's also a message indicating the Python signature status is;
> Resource or signature modified

A further message states;
> “Visual Studio Code” must be signed by registered developer or by Apple
		“python” must have executable checksum “F71676143C…”

Hopefully this helps and the devs can sort this out because right now, short of turning off Little Snitch, I can't use PIO at all.

demosthien, 2019-05-31:
If the recent issues submitted are anything to go by that URL code issue seems to be an ongoing one.

demosthien, 2019-07-13:
Soooo... No comment on this issue at all? Looking at the list of issues it seems to be a fairly common problem for a lot of users.

ivankravets, 2019-07-13:
Please forward to https://github.com/microsoft/vscode

closed 2019-07-13 by ivankravets