platformio/platformio-vscode-ide #811: Installation%20Manager

ccpcatarino, 2019-05-30:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.34.0%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Windows_NT%2C%2010.0.18362%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20PIP%20Core%3A%20If%20you%20have%20antivirus%2Ffirewall%2Fdefender%20software%20in%20a%20system%2C%20try%20to%20disable%20it%20for%20a%20while.%20%0A%20%20%20Retrying%20(Retry(total%3D4%2C%20connect%3DNone%2C%20read%3DNone%2C%20redirect%3DNone%2C%20status%3DNone))%20after%20connection%20broken%20by%20'ProtocolError('Connection%20aborted.'%2C%20error(10054%2C%20'Uma%20liga%5Cxe7%5Cxe3o%20existente%20foi%20for%5Cxe7ada%20a%20fechar%20pelo%20anfitri%5Cxe3o%20remoto'))'%3A%20%2Fsimple%2Fplatformio%2F%0D%0A%20%20Retrying%20(Retry(total%3D3%2C%20connect%3DNone%2C%20read%3DNone%2C%20redirect%3DNone%2C%20status%3DNone))%20after%20connection%20broken%20by%20'ProtocolError('Connection%20aborted.'%2C%20error(10054%2C%20'Uma%20liga%5Cxe7%5Cxe3o%20existente%20foi%20for%5Cxe7ada%20a%20fechar%20pelo%20anfitri%5Cxe3o%20remoto'))'%3A%20%2Fsimple%2Fplatformio%2F%0D%0A%20%20Retrying%20(Retry(total%3D2%2C%20connect%3DNone%2C%20read%3DNone%2C%20redirect%3DNone%2C%20status%3DNone))%20after%20connection%20broken%20by%20'ProtocolError('Connection%20aborted.'%2C%20error(10054%2C%20'Uma%20liga%5Cxe7%5Cxe3o%20existente%20foi%20for%5Cxe7ada%20a%20fechar%20pelo%20anfitri%5Cxe3o%20remoto'))'%3A%20%2Fsimple%2Fplatformio%2F%0D%0A%20%20Retrying%20(Retry(total%3D1%2C%20connect%3DNone%2C%20read%3DNone%2C%20redirect%3DNone%2C%20status%3DNone))%20after%20connection%20broken%20by%20'ProtocolError('Connection%20aborted.'%2C%20error(10054%2C%20'Uma%20liga%5Cxe7%5Cxe3o%20existente%20foi%20for%5Cxe7ada%20a%20fechar%20pelo%20anfitri%5Cxe3o%20remoto'))'%3A%20%2Fsimple%2Fplatformio%2F%0D%0A%20%20Retrying%20(Retry(total%3D0%2C%20connect%3DNone%2C%20read%3DNone%2C%20redirect%3DNone%2C%20status%3DNone))%20after%20connection%20broken%20by%20'ProtocolError('Connection%20aborted.'%2C%20error(10054%2C%20'Uma%20liga%5Cxe7%5Cxe3o%20existente%20foi%20for%5Cxe7ada%20a%20fechar%20pelo%20anfitri%5Cxe3o%20remoto'))'%3A%20%2Fsimple%2Fplatformio%2F%0D%0A%20%20Retrying%20(Retry(total%3D4%2C%20connect%3DNone%2C%20read%3DNone%2C%20redirect%3DNone%2C%20status%3DNone))%20after%20connection%20broken%20by%20'ProtocolError('Connection%20aborted.'%2C%20error(10054%2C%20'Uma%20liga%5Cxe7%5Cxe3o%20existente%20foi%20for%5Cxe7ada%20a%20fechar%20pelo%20anfitri%5Cxe3o%20remoto'))'%3A%20%2Fpackages%2Ffe%2F01%2F69aa7d8ef8cd74493338396ff86dc1bbfe85ae58b77fc705924c920a38eb%2Fplatformio-3.6.7-py27-none-any.whl%0D%0A%20%20Retrying%20(Retry(total%3D3%2C%20connect%3DNone%2C%20read%3DNone%2C%20redirect%3DNone%2C%20status%3DNone))%20after%20connection%20broken%20by%20'ProtocolError('Connection%20aborted.'%2C%20error(10054%2C%20'Uma%20liga%5Cxe7%5Cxe3o%20existente%20foi%20for%5Cxe7ada%20a%20fechar%20pelo%20anfitri%5Cxe3o%20remoto'))'%3A%20%2Fpackages%2Ffe%2F01%2F69aa7d8ef8cd74493338396ff86dc1bbfe85ae58b77fc705924c920a38eb%2Fplatformio-3.6.7-py27-none-any.whl%0D%0A%20%20Retrying%20(Retry(total%3D2%2C%20connect%3DNone%2C%20read%3DNone%2C%20redirect%3DNone%2C%20status%3DNone))%20after%20connection%20broken%20by%20'ProtocolError('Connection%20aborted.'%2C%20error(10054%2C%20'Uma%20liga%5Cxe7%5Cxe3o%20existente%20foi%20for%5Cxe7ada%20a%20fechar%20pelo%20anfitri%5Cxe3o%20remoto'))'%3A%20%2Fpackages%2Ffe%2F01%2F69aa7d8ef8cd74493338396ff86dc1bbfe85ae58b77fc705924c920a38eb%2Fplatformio-3.6.7-py27-none-any.whl%0D%0A%20%20Retrying%20(Retry(total%3D1%2C%20connect%3DNone%2C%20read%3DNone%2C%20redirect%3DNone%2C%20status%3DNone))%20after%20connection%20broken%20by%20'ProtocolError('Connection%20aborted.'%2C%20error(10054%2C%20'Uma%20liga%5Cxe7%5Cxe3o%20existente%20foi%20for%5Cxe7ada%20a%20fechar%20pelo%20anfitri%5Cxe3o%20remoto'))'%3A%20%2Fpackages%2Ffe%2F01%2F69aa7d8ef8cd74493338396ff86dc1bbfe85ae58b77fc705924c920a38eb%2Fplatformio-3.6.7-py27-none-any.whl%0D%0A%20%20Retrying%20(Retry(total%3D0%2C%20connect%3DNone%2C%20read%3DNone%2C%20redirect%3DNone%2C%20status%3DNone))%20after%20connection%20broken%20by%20'ProtocolError('Connection%20aborted.'%2C%20error(10054%2C%20'Uma%20liga%5Cxe7%5Cxe3o%20existente%20foi%20for%5Cxe7ada%20a%20fechar%20pelo%20anfitri%5Cxe3o%20remoto'))'%3A%20%2Fpackages%2Ffe%2F01%2F69aa7d8ef8cd74493338396ff86dc1bbfe85ae58b77fc705924c920a38eb%2Fplatformio-3.6.7-py27-none-any.whl%0D%0ACould%20not%20install%20packages%20due%20to%20an%20EnvironmentError%3A%20HTTPSConnectionPool(host%3D'files.pythonhosted.org'%2C%20port%3D443)%3A%20Max%20retries%20exceeded%20with%20url%3A%20%2Fpackages%2Ffe%2F01%2F69aa7d8ef8cd74493338396ff86dc1bbfe85ae58b77fc705924c920a38eb%2Fplatformio-3.6.7-py27-none-any.whl%20(Caused%20by%20ProtocolError('Connection%20aborted.'%2C%20error(10054%2C%20'Uma%20liga%5Cxe7%5Cxe3o%20existente%20foi%20for%5Cxe7ada%20a%20fechar%20pelo%20anfitri%5Cxe3o%20remoto')))%0D%0A%0D%0A%0A%09at%20f.value.i.runCommand%20(c%3A%5CUsers%5Cccpin%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A26495)%0A%09at%20ChildProcess.h%20(c%3A%5CUsers%5Cccpin%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A5904)%0A%09at%20ChildProcess.emit%20(events.js%3A182%3A13)%0A%09at%20ChildProcess.cp.emit%20(c%3A%5CUsers%5Cccpin%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Ccross-spawn%5Clib%5Cenoent.js%3A34%3A29)%0A%09at%20maybeClose%20(internal%2Fchild_process.js%3A961%3A16)%0A%09at%20Process.ChildProcess._handle.onexit%20(internal%2Fchild_process.js%3A248%3A5)%0A%60%60%60%0A

demosthien, 2019-05-31:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.34.0
PIO IDE: v1.7.1
System: Windows_NT, 10.0.18362, x64

# Exception
```
Error: PIP Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while. 
   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', error(10054, 'Uma liga\xe7\xe3o existente foi for\xe7ada a fechar pelo anfitri\xe3o remoto'))': /simple/platformio/
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', error(10054, 'Uma liga\xe7\xe3o existente foi for\xe7ada a fechar pelo anfitri\xe3o remoto'))': /simple/platformio/
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', error(10054, 'Uma liga\xe7\xe3o existente foi for\xe7ada a fechar pelo anfitri\xe3o remoto'))': /simple/platformio/
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', error(10054, 'Uma liga\xe7\xe3o existente foi for\xe7ada a fechar pelo anfitri\xe3o remoto'))': /simple/platformio/
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', error(10054, 'Uma liga\xe7\xe3o existente foi for\xe7ada a fechar pelo anfitri\xe3o remoto'))': /simple/platformio/
  Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', error(10054, 'Uma liga\xe7\xe3o existente foi for\xe7ada a fechar pelo anfitri\xe3o remoto'))': /packages/fe/01/69aa7d8ef8cd74493338396ff86dc1bbfe85ae58b77fc705924c920a38eb/platformio-3.6.7-py27-none-any.whl
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', error(10054, 'Uma liga\xe7\xe3o existente foi for\xe7ada a fechar pelo anfitri\xe3o remoto'))': /packages/fe/01/69aa7d8ef8cd74493338396ff86dc1bbfe85ae58b77fc705924c920a38eb/platformio-3.6.7-py27-none-any.whl
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', error(10054, 'Uma liga\xe7\xe3o existente foi for\xe7ada a fechar pelo anfitri\xe3o remoto'))': /packages/fe/01/69aa7d8ef8cd74493338396ff86dc1bbfe85ae58b77fc705924c920a38eb/platformio-3.6.7-py27-none-any.whl
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', error(10054, 'Uma liga\xe7\xe3o existente foi for\xe7ada a fechar pelo anfitri\xe3o remoto'))': /packages/fe/01/69aa7d8ef8cd74493338396ff86dc1bbfe85ae58b77fc705924c920a38eb/platformio-3.6.7-py27-none-any.whl
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', error(10054, 'Uma liga\xe7\xe3o existente foi for\xe7ada a fechar pelo anfitri\xe3o remoto'))': /packages/fe/01/69aa7d8ef8cd74493338396ff86dc1bbfe85ae58b77fc705924c920a38eb/platformio-3.6.7-py27-none-any.whl
Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Max retries exceeded with url: /packages/fe/01/69aa7d8ef8cd74493338396ff86dc1bbfe85ae58b77fc705924c920a38eb/platformio-3.6.7-py27-none-any.whl (Caused by ProtocolError('Connection aborted.', error(10054, 'Uma liga\xe7\xe3o existente foi for\xe7ada a fechar pelo anfitri\xe3o remoto')))

	at f.value.i.runCommand (c:\Users\ccpin\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:26495)
	at ChildProcess.h (c:\Users\ccpin\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:5904)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (c:\Users\ccpin\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

ivankravets, 2019-07-13:
> Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Max retries exceeded with url: [...]

Please check your Internet connection

closed 2019-07-13 by ivankravets