platformio/platformio-vscode-ide #812: Installation%20Manager

dwaite, 2019-05-30:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.34.0%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Darwin%2C%2018.6.0%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20Could%20not%20create%20PIO%20Core%20Virtual%20Environment.%20Please%20create%20it%20manually%20-%3E%20http%3A%2F%2Fbit.ly%2Fpio-core-virtualenv%20%0A%20Error%3A%20Failed%20to%20download%20file%20https%3A%2F%2Ffiles.pythonhosted.org%2Fpackages%2F4e%2F8b%2F75469c270ac544265f0020aa7c4ea925c5284b23e445cf3aa8b99f662690%2Fvirtualenv-16.1.0.tar.gz%3A%20%0A%09at%20t.value%20(%2FUsers%2Fdavid%2F.vscode%2Fextensions%2Fplatformio.platformio-ide-1.7.1%2Fnode_modules%2Fplatformio-node-helpers%2Fdist%2Findex.js%3A1%3A25605)%0A%60%60%60%0A

demosthien, 2019-05-31:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.34.0
PIO IDE: v1.7.1
System: Darwin, 18.6.0, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Failed to download file https://files.pythonhosted.org/packages/4e/8b/75469c270ac544265f0020aa7c4ea925c5284b23e445cf3aa8b99f662690/virtualenv-16.1.0.tar.gz: 
	at t.value (/Users/david/.vscode/extensions/platformio.platformio-ide-1.7.1/node_modules/platformio-node-helpers/dist/index.js:1:25605)
```

ivankravets, 2019-07-13:
>  Error: Failed to download file https://files.pythonhosted.org/packages/4e/8b/75469c270ac544265f0020aa7c4ea925c5284b23e445cf3aa8b99f662690/virtualenv-16.1.0.tar.gz: 

Please check your internet connection

closed 2019-07-13 by ivankravets