platformio/platformio-vscode-ide #817: Installation%20Manager

megadur, 2019-05-31:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.34.0%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Windows_NT%2C%2010.0.17763%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20Could%20not%20create%20PIO%20Core%20Virtual%20Environment.%20Please%20create%20it%20manually%20-%3E%20http%3A%2F%2Fbit.ly%2Fpio-core-virtualenv%20%0A%20Error%3A%20EPERM%3A%20operation%20not%20permitted%2C%20mkdir%20'C%3A%5CProgram%20Files%5CMicrosoft%20VS%20Code%5C%25HOMEPATH%25'%0A%09at%20t.value%20(C%3A%5CUsers%5CDonTillo%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A25605)%0A%60%60%60%0A

demosthien, 2019-05-31:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.34.0
PIO IDE: v1.7.1
System: Windows_NT, 10.0.17763, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: EPERM: operation not permitted, mkdir 'C:\Program Files\Microsoft VS Code\%HOMEPATH%'
	at t.value (C:\Users\DonTillo\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:25605)
```

ivankravets, 2019-07-13:
Did you override %USERPROFILE% environment variable?

closed 2019-07-13 by ivankravets