platformio/platformio-vscode-ide #819: Start%20PIO%20Home%20Server

cyrusbuilt, 2019-05-31:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.34.0%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Darwin%2C%2018.5.0%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20Traceback%20(most%20recent%20call%20last)%3A%0A%20%20File%20%22%3Cstring%3E%22%2C%20line%201%2C%20in%20%3Cmodule%3E%0A%20%20File%20%22%3Cstring%3E%22%2C%20line%2038%2C%20in%20%3Cmodule%3E%0A%20%20File%20%22%3Cstring%3E%22%2C%20line%2039%2C%20in%20%3Cmodule%3E%0A%20%20File%20%22%3Cstring%3E%22%2C%20line%2018%2C%20in%20load_module%0A%20%20File%20%22%3Cstring%3E%22%2C%20line%2023%2C%20in%20%3Cmodule%3E%0AImportError%3A%20cannot%20import%20name%20get_project_test_dir%0A%0A%09at%20P.Promise%20(%2FUsers%2Fsdsdev%2F.vscode%2Fextensions%2Fplatformio.platformio-ide-1.7.1%2Fnode_modules%2Fplatformio-node-helpers%2Fdist%2Findex.js%3A1%3A15270)%0A%09at%20ChildProcess.h%20(%2FUsers%2Fsdsdev%2F.vscode%2Fextensions%2Fplatformio.platformio-ide-1.7.1%2Fnode_modules%2Fplatformio-node-helpers%2Fdist%2Findex.js%3A1%3A5904)%0A%09at%20ChildProcess.emit%20(events.js%3A182%3A13)%0A%09at%20maybeClose%20(internal%2Fchild_process.js%3A961%3A16)%0A%09at%20Process.ChildProcess._handle.onexit%20(internal%2Fchild_process.js%3A248%3A5)%0A%60%60%60%0A

cyrusbuilt, 2019-05-31:
Crap. That is the report text it generated. Let me see if I can clean it up.

cyrusbuilt, 2019-05-31:
When clicking "Open" under "PIO Home" or pretty much any other option under "PIO Home", I get this:

Error: Traceback (most recent call last): File "<string>", line 1, in <module> File "<string>", line 38, in <module> File "<string>", line 39, in <module> File "<string>", line 18, in load_module File "<string>", line 23, in <module> ImportError: cannot import name get_project_test_dir at P.Promise (/Users/sdsdev/.vscode/extensions/platformio.platformio-ide-1.7.1/node_modules/platformio-node-helpers/dist/index.js:1:15270) at ChildProcess.h (/Users/sdsdev/.vscode/extensions/platformio.platformio-ide-1.7.1/node_modules/platformio-node-helpers/dist/index.js:1:5904) at ChildProcess.emit (events.js:182:13) at maybeClose (internal/child_process.js:961:16) at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)

carmbrust, 2019-05-31:
I'm seeing this also. Not sure what changed??

carmbrust, 2019-05-31:
So, I did an upgrade platformio to 4.0.0a16 and things are working again (i.e. PIO Home page opens correctly and the above message is not generated).

ivankravets, 2019-07-13:
Please open PlatformIO IDE Terminal and type
```
pio upgrade
```

Restart VSCode.

closed 2019-07-13 by ivankravets