platformio/platformio-vscode-ide #820: Installation%20Manager

eliott-jouneau, 2019-05-31:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.34.0%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Windows_NT%2C%2010.0.17134%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20Could%20not%20create%20PIO%20Core%20Virtual%20Environment.%20Please%20create%20it%20manually%20-%3E%20http%3A%2F%2Fbit.ly%2Fpio-core-virtualenv%20%0A%20Error%3A%20Virtualenv%20Create%3A%20Traceback%20(most%20recent%20call%20last)%3A%0D%0A%20%20File%20%22C%3A%5CUsers%5Cstarl%5C.platformio%5C.cache%5Ctmp-38624ojLPx4VvNX9f%5Cvirtualenv-16.1.0%5Csrc%5Cvirtualenv.py%22%2C%20line%202462%2C%20in%20%3Cmodule%3E%0D%0A%20%20%20%20main()%0D%0A%20%20File%20%22C%3A%5CUsers%5Cstarl%5C.platformio%5C.cache%5Ctmp-38624ojLPx4VvNX9f%5Cvirtualenv-16.1.0%5Csrc%5Cvirtualenv.py%22%2C%20line%20762%2C%20in%20main%0D%0A%20%20%20%20symlink%3Doptions.symlink%2C%0D%0A%20%20File%20%22C%3A%5CUsers%5Cstarl%5C.platformio%5C.cache%5Ctmp-38624ojLPx4VvNX9f%5Cvirtualenv-16.1.0%5Csrc%5Cvirtualenv.py%22%2C%20line%20998%2C%20in%20create_environment%0D%0A%20%20%20%20install_python(home_dir%2C%20lib_dir%2C%20inc_dir%2C%20bin_dir%2C%20site_packages%3Dsite_packages%2C%20clear%3Dclear%2C%20symlink%3Dsymlink)%0D%0A%20%20File%20%22C%3A%5CUsers%5Cstarl%5C.platformio%5C.cache%5Ctmp-38624ojLPx4VvNX9f%5Cvirtualenv-16.1.0%5Csrc%5Cvirtualenv.py%22%2C%20line%201317%2C%20in%20install_python%0D%0A%20%20%20%20shutil.copyfile(executable%2C%20py_executable)%0D%0A%20%20File%20%22C%3A%5CUsers%5Cstarl%5C.platformio%5Cpython27%5Clib%5Cshutil.py%22%2C%20line%2083%2C%20in%20copyfile%0D%0A%20%20%20%20with%20open(dst%2C%20'wb')%20as%20fdst%3A%0D%0AIOError%3A%20%5BErrno%2013%5D%20Permission%20denied%3A%20'C%3A%5C%5CUsers%5C%5Cstarl%5C%5C.platformio%5C%5Cpenv%5C%5CScripts%5C%5Cpython.exe'%0D%0A%0ANew%20python%20executable%20in%20C%3A%5CUsers%5Cstarl%5C.platformio%5Cpenv%5CScripts%5Cpython.exe%0D%0A%0A%09at%20t.value%20(C%3A%5CUsers%5Cstarl%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A25605)%0A%60%60%60%0A

demosthien, 2019-05-31:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.34.0
PIO IDE: v1.7.1
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\starl\.platformio\.cache\tmp-38624ojLPx4VvNX9f\virtualenv-16.1.0\src\virtualenv.py", line 2462, in <module>
    main()
  File "C:\Users\starl\.platformio\.cache\tmp-38624ojLPx4VvNX9f\virtualenv-16.1.0\src\virtualenv.py", line 762, in main
    symlink=options.symlink,
  File "C:\Users\starl\.platformio\.cache\tmp-38624ojLPx4VvNX9f\virtualenv-16.1.0\src\virtualenv.py", line 998, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\starl\.platformio\.cache\tmp-38624ojLPx4VvNX9f\virtualenv-16.1.0\src\virtualenv.py", line 1317, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\starl\.platformio\python27\lib\shutil.py", line 83, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: 'C:\\Users\\starl\\.platformio\\penv\\Scripts\\python.exe'

New python executable in C:\Users\starl\.platformio\penv\Scripts\python.exe

	at t.value (C:\Users\starl\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:25605)
```

LoneGunnerMe, 2019-06-02:
You can open a new terminal and run the following command:
pio upgrade --dev
pio update

ivankravets, 2019-07-13:
It seems like your antivirus blocks PlatformIO installer. See http://bit.ly/pio-core-virtualenv

closed 2019-07-13 by ivankravets