platformio/platformio-vscode-ide #822: Installation%20Manager

RuhollahArabi, 2019-06-01:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.34.0%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Darwin%2C%2018.5.0%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20PIP%20Core%3A%20%2FUsers%2Fiamir%2F.platformio%2Fpenv%2Fbin%2Fpython2.7%3A%20No%20module%20named%20pip%0A%0A%09at%20f.value.i.runCommand%20(%2FUsers%2Fiamir%2F.vscode%2Fextensions%2Fplatformio.platformio-ide-1.7.1%2Fnode_modules%2Fplatformio-node-helpers%2Fdist%2Findex.js%3A1%3A26495)%0A%09at%20ChildProcess.h%20(%2FUsers%2Fiamir%2F.vscode%2Fextensions%2Fplatformio.platformio-ide-1.7.1%2Fnode_modules%2Fplatformio-node-helpers%2Fdist%2Findex.js%3A1%3A5904)%0A%09at%20ChildProcess.emit%20(events.js%3A182%3A13)%0A%09at%20maybeClose%20(internal%2Fchild_process.js%3A961%3A16)%0A%09at%20Process.ChildProcess._handle.onexit%20(internal%2Fchild_process.js%3A248%3A5)%0A%60%60%60%0A

LoneGunnerMe, 2019-06-02:
You can open a new terminal and run the following command:
pio upgrade --dev
pio update

ivankravets, 2019-07-13:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/815

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

closed 2019-07-13 by ivankravets