platformio/platformio-vscode-ide #823: Debugger: Memory viewer shows 0x00 everywhere for specific bytes to read

mirao, 2019-06-01:
**Steps to reproduce:**

- Debug an app on a device using F5. (the device I tested is STM32F407G-DISC1)
- Suppose value 0x22 is stored at the address 0x2001fff6
- Open Debug => MEMORY
- Click on "+"
- Set start memory address 0x2001fff6
- Set 10 bytes
- You can see the value 0x22 (and also some other non zero bytes). It's correct:
[image]
- Set start memory address 0x2001fff6 again
- Set 11 bytes (or other value greater than 10)

**Expected result**
- You can see 0x22 at the address 0x2001fff6

**Actual result**
- You can see 0x00 at the address 0x2001fff6
- Also bytes at other addresses, such as 0x2001fffc or 0x2001fffd are 0x00 which is incorrect (compare with previous screenshot):
[image]

**PIO config**
```ini
[env:disco_f407vg]
platform = ststm32
board = disco_f407vg
framework = cmsis
```
**Used SW:**
- PlatformIO, version 3.6.7
- VSCode 1.34.0
- Ubuntu 18.10

**Used HW:**
- STM32F407G-DISC1

valeros, 2019-07-16:
Hi @mirao! I suppose it's correct behavior as STM32F407VG has 192K of RAM that is organized in two banks:
```
RAM (xrw)        : ORIGIN = 0x20000000, LENGTH = 128K
CCMRAM (rw)      : ORIGIN = 0x10000000, LENGTH = 64K
```
So memory viewer fails to read memory segment outside RAM space (`0x2001fff6 + 11 =  0x20020001` ).

mirao, 2019-07-16:
OK, thanks for explanation. I would expect that bytes below `0x20020000` are always shown correctly even if `start + offset` goes outside RAM. And only >=`0x20020000` should be hidden (or shown as virtual `00` maybe).
Is it possible to show bytes specified by top address and negative offset? E.g. "show me 11 bytes below `0x20020000`". It's impractical to compute start address so that end address doesn't overflow top.

ivankravets, 2019-07-23:
I'm not sure that we can fix this issue. You ask via VSCode UI to read some block of memory. We transfer this request to the underlying debug client. We can not control this.

The latest version reads data from memory splitting requests to 1Kb blocks. It should help with this issue.

closed 2019-07-23 by ivankravets