platformio/platformio-vscode-ide #824: Installation%20Manager

reyking, 2019-06-03:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.34.0%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Windows_NT%2C%2010.0.17763%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20Could%20not%20create%20PIO%20Core%20Virtual%20Environment.%20Please%20create%20it%20manually%20-%3E%20http%3A%2F%2Fbit.ly%2Fpio-core-virtualenv%20%0A%20TypeError%20%5BERR_INVALID_ARG_TYPE%5D%3A%20The%20%22path%22%20argument%20must%20be%20of%20type%20string.%20Received%20type%20undefined%0A%09at%20t.value%20(c%3A%5CUsers%5Chansr%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A25605)%0A%60%60%60%0A

pfeerick, 2019-06-08:
Don't know why github decided to urlencode the issue making it ... hard to read! :-/

# Configuration
VSCode: 1.34.0
PIO IDE: v1.7.1
System: Windows_NT, 10.0.17763, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received type undefined
	at t.value (c:\Users\hansr\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:25605)
```

ivankravets, 2019-07-13:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/750

closed 2019-07-13 by ivankravets