platformio/platformio-vscode-ide #837: Error: Detected unknown package 'tool-pioplus'

ABOSTM, 2019-06-14:
Hi,
I installed PlatfomIO IDE (v1.7.1) with VS code (v1.35.1) on Windows 10.
Installation seems good, but I face error while using it.
For example:
> pio home
CorePackageManager: Installing tool-pioplus @ ^2.1.4
Error: Detected unknown package 'tool-pioplus'

I use proxy, and I configured environment variables like that:
HTTP_PROXY = http://login:password@proxyIP:proxyPort/
HTTPS_PROXY = http://login:password@proxyIP:proxyPort/
When running echo %HTTP_PROXY% from system terminal, I get the right values (like above)

In VScode menu help->Toggle Developper tools -> Console
I found following error:
Error: Error: Detected unknown package 'tool-pioplus'
	at P.Promise (C:\Users\frq07334\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:15270)
	at ChildProcess.h (C:\Users\frq07334\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:5904)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (C:\Users\frq07334\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)

It looks similar to other issue reported, but I can't get the right solution for my case.

I deactivated AntiVirus
I tried to close/ open Vs code
I tryed to uninstall and reinstall platformio IDE
I tried to remove c:\Users\username\.platformio
I looked at Windows task manager, but could not found pioplus.exe processes.

Any help is welcome,
Thanks

pfeerick, 2019-07-13:
Any progress with this? Any change now that PIOv4 out? 

You could try see if downloading it manually, and unpacking the contents into `%userprofile%\.platformio\packages\tool-pioplus` works... (You may need 7zip in order to unpack the download package).

https://dl.bintray.com/platformio/dl-packages/tool-pioplus-windows_amd64-2.5.2.tar.gz
or 
https://dl.bintray.com/platformio/dl-packages/tool-pioplus-windows_x86-2.5.2.tar.gz

ivankravets, 2019-07-13:
Please upgrade PIO Core to 4.0, there is no need in this `pioplus` now:
```
pio upgrade
```

in PIO IDE Temrinal.

ABOSTM, 2019-07-15:
Hi, 
Thank you both for your answers.
I tried both solutions : Install manually tool-pioplus and upgrade pio core to 4.0
But unfortunately I am now facing a new error (see below).
I tried to uninsatll and reinstall platformio extension without success.

Error: Error: Detected unknown package 'contrib-pysite'

	at P.Promise (C:\Users\frq07334\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:15270)
	at ChildProcess.h (C:\Users\frq07334\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:5904)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (C:\Users\frq07334\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)

ivankravets, 2019-07-15:
Do you have antivirus tools in the system? They can block PlatformIO. Also, check your firewall.

ABOSTM, 2019-07-15:
:-(    I try to deacivate antivirus and firewall, but it doesn't help
(and even if it would work, deactivating antivirus would not have been an acceptable solution for me).
I tried to install contrib-pysite manually :
https://dl.bintray.com/platformio/dl-packages/contrib-pysite-windows_x86-2.37.190418.tar.gz
in %userprofile%\.platformio\packages\contrib-pysite
But still the same issue reported

pfeerick, 2019-07-15:
If it had been identified as an antivirus issue, you could have put in place an exception so it behaves, but that doesn't look to be the case.

Best bet now is to probably delete (or rename, if you want to keep it just in case things go terribly, terribly wrong) the entire `%userprofile%.platformio` folder, and let the VSCode PlatformIO extension try to do a clean install of the current version.

ABOSTM, 2019-07-16:
I removed %userprofile%.platformio and reinstall extension.
It still miss contrib-pysite.
I try again to install it manually, but always the same error
Error: Error: Detected unknown package 'contrib-pysite' ...

It looks like my manuall installation is inefficient.
Note: I am on Intel 64 processor but I couldn't found corresponding version, thus I installed https://dl.bintray.com/platformio/dl-packages/contrib-pysite-windows_x86-2.37.190418.tar.gz

pfeerick, 2019-07-16:
At what point does the 'unknown package' error come up? ie. After installing the extension, but before restarting vscode or after restarting vscode or doing a build or something else? As I'll see if I can reproduce at least the breakage...

ABOSTM, 2019-07-16:
Error doesn't appear at reinstallation, 
but it appears after restart of VScode.
It appears as a pop up after a while (about 20-30 secondes) doing nothing.

I also have an error message when running command 'pio home' in pio terminal (Platformio -> Miscellaneous -> New Terminal)

ivankravets, 2019-07-16:
Please open PlatformIO IDE Terminal and type
```
python -c "import requests; print(requests.get('https://dl.bintray.com/platformio/dl-packages/manifest.json').json())"
```

What do you see?

ABOSTM, 2019-07-16:
Here it is:
python -c "import requests; print(requests.get('https://dl.bintray.com/platformio/dl-packages/manifest.json').json())"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\frq07334\.platformio\penv\lib\site-packages\requests\api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "C:\Users\frq07334\.platformio\penv\lib\site-packages\requests\api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "C:\Users\frq07334\.platformio\penv\lib\site-packages\requests\sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "C:\Users\frq07334\.platformio\penv\lib\site-packages\requests\sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "C:\Users\frq07334\.platformio\penv\lib\site-packages\requests\adapters.py", line 514, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: HTTPSConnectionPool(host='dl.bintray.com', port=443): Max retries exceeded with url: /platformio/dl-packages/manifest.json (Caused by SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certifi SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:661)'),))

pfeerick, 2019-07-16:
Since you're behind a proxy, try adding the proxy configuration... i.e.
```
python -c "import requests; print(requests.get('https://dl.bintray.com/platformio/dl-packages/manifest.json',proxies={'https':'https://proxyip:proxyport'}).json())"
```

However, I'm wondering if this is one of those nasty python3 issues... i.e. https://github.com/bloomreach/s4cmd/issues/111#issuecomment-406839514

ABOSTM, 2019-07-16:
I was expecting platformio could retrieve proxy from VScode or Environment variable (like VScode do).
Whatever, adding proxy configuration in the command, I get new proxy error below.
About Python, I am using Python2.7.13 not Python3.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\frq07334\.platformio\penv\lib\site-packages\requests\api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "C:\Users\frq07334\.platformio\penv\lib\site-packages\requests\api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "C:\Users\frq07334\.platformio\penv\lib\site-packages\requests\sessions.py", line 524, in request
    prep.url, proxies, stream, verify, cert
  File "C:\Users\frq07334\.platformio\penv\lib\site-packages\requests\sessions.py", line 699, in merge_environment_settings
    no_proxy = proxies.get('no_proxy') if proxies is not None else None
AttributeError: 'set' object has no attribute 'get'

ivankravets, 2019-07-16:
PlatformIO does this. See http://docs.platformio.org/en/latest/ide/vscode.html#proxy-server-support

Do you use new terminal via clicking on Terminal icon in the bottom status bar?

ABOSTM, 2019-07-16:
I use platformIO menu  -> quick access -> Miscellaneous -> New terminal
and I get a windows with this content:
          Windows PowerShell
          Copyright (C) Microsoft Corporation. All rights reserved.

           PS C:\Arduino\arduino-1.8.9\portable\packages\STM32\hardware\stm32\1.6.0>

I don't have terminal icon in status bar. I only have an house icon for 'PlatformIO: Home'

ivankravets, 2019-07-16:
```
python -c "import os; print(os.environ)"
```

Do you see proxy settings in env?

ABOSTM, 2019-07-16:
Yes I can see HTTP_PROXY and HTTPS_PROXY defined

ivankravets, 2019-07-16:
```
python -c "import requests; print(requests.get('https://dl.bintray.com/platformio/dl-packages/manifest.json', verify=False).json())"
```

Does it work now?

ivankravets, 2019-07-16:
Related https://stackoverflow.com/questions/15445981/how-do-i-disable-the-security-certificate-check-in-python-requests

ivankravets, 2019-07-16:
Did you disable "Strict SSL" in VSCode?

ABOSTM, 2019-07-16:
Yes I disabled "Strict SSL" in VSCode.
There is one option in VScode which I am not sure: 'http: Proxy support' which is 'on'.
but whatever setting I get an error (which may be different depending on the setting)

PS C:\Arduino\arduino-1.8.9\portable\packages\STM32\hardware\stm32\1.6.0> python -c "import requests; print(requests.get('https://dl.bintray.com/platformio/dl-packages/manifest.json', verify=False).json())"
C:\Python27\lib\site-packages\urllib3\connectionpool.py:851: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
  InsecureRequestWarning)
{u'framework-shakti-sdk': [{u'url': u'https://dl.bintray.com/platformio/dl-packages/framework-shakti-sdk-0.0.190715.tar.gz', u'sha1': u'44f4c952d9315c9b60e9581e4b3222aa895cce7e', u'system': u'*', u'version': u'0.0.190715'}], u'framework-arduinoststm8': [{u'url': u'https://dl.bintray.com/platformio/dl-packages/framework-arduinoststm8-0.40.181216.tar.gz', u'sha1': u'7f91214cf9eac9814862f47ee54047c1673b9f86', u'system': u'*', u'version': u'0.40.181216'}], u'tool-stm32duino': [{u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stm32duino-darwin_x86_64-1.0.1.tar.gz', u'sha1': u'ea4f196bfb4bbdd7cd610b047c16a60b1f5ba4c5', u'system': [u'darwin_x86_64', u'darwin_i386'], u'version': u'1.0.1'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stm32duino-linux_armv6l-1.0.1.tar.gz', u'sha1': u'd7d96b7fc9d454cc3ff102248bea89ba05fb5778', u'system': [u'linux_armv6l', u'linux_armv7l', u'linux_armv8l', u'linux_aarch64'], u'version': u'1.0.1'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stm32duino-linux_i686-1.0.1.tar.gz', u'sha1': u'83b0a0885b4e31da79a352ea39193136c17f4fc1', u'system': [u'linux_i686'], u'version': u'1.0.1'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stm32duino-linux_x86_64-1.0.1.tar.gz', u'sha1': u'874dd34a69872e8b0c42dd3fd5a00d17a8ba56b9', u'system': [u'linux_x86_64'], u'version': u'1.0.1'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stm32duino-windows-1.0.1.tar.gz', u'sha1': u'373e5a3dc716362845501d254aac93e84a2dcbc4', u'system': [u'windows', u'windows_amd64', u'windows_x86'], u'version': u'1.0.1'}], u'tool-stlink': [{u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stlink-darwin_x86_64-1.10200.0.tar.gz', u'sha1': u'28357a1d7e9defa7bdad5fb3472415d8c31b07be', u'system':
[u'darwin_x86_64', u'darwin_i386'], u'version': u'1.10200.0'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stlink-darwin_x86_64-1.10301.1.tar.gz', u'sha1': u'ed1d1d42dc3d0141311956b89b9716d8eb33acd9', u'system': [u'darwin_x86_64', u'darwin_i386'], u'version': u'1.10301.1'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stlink-darwin_x86_64-1.10400.0.tar.gz', u'sha1': u'3f0dc828cc1590d1bc1c773114419a497a4dc55d', u'system': [u'darwin_x86_64', u'darwin_i386'], u'version': u'1.10400.0'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stlink-linux_armv6l-1.10200.0.tar.gz', u'sha1': u'ceac11c6bb5a9a366c5ca87bf986d540e1842ac5', u'system': [u'linux_armv6l', u'linux_armv7l', u'linux_armv8l', u'linux_aarch64'], u'version': u'1.10200.0'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stlink-linux_armv6l-1.10301.0.tar.gz', u'sha1': u'51031318d4c1579a297447849a0b7ed80f062864', u'system': [u'linux_armv6l', u'linux_armv7l', u'linux_armv8l', u'linux_aarch64'], u'version': u'1.10301.0'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stlink-linux_armv6l-1.10400.0.tar.gz', u'sha1': u'636fbab0d8cb40dc87cf5be70a4d5a96c2a15de7', u'system': [u'linux_armv6l', u'linux_armv7l', u'linux_armv8l', u'linux_aarch64'], u'version': u'1.10400.0'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stlink-linux_i686-1.10200.0.tar.gz', u'sha1': u'4584cdbb4bfc9323b975a8a92980959058bc81ad', u'system': [u'linux_i686'], u'version': u'1.10200.0'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stlink-linux_i686-1.10301.1.tar.gz', u'sha1': u'd2cf8158ea91e503d9c34578d1737d165de012a3', u'system': u'linux_i686', u'version': u'1.10301.1'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stlink-linux_i686-1.10400.0.tar.gz', u'sha1': u'91fe43164ce31edc6cf21121d65c32061eabcd7e', u'system': u'linux_i686', u'version': u'1.10400.0'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stlink-linux_x86_64-1.10200.0.tar.gz', u'sha1': u'25b01ce0d4825a112cb1b1fa2dbc4de59656f825', u'system': [u'linux_x86_64'], u'version': u'1.10200.0'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stlink-linux_x86_64-1.10301.1.tar.gz', u'sha1': u'a2d512419479009702e81d7de34010399c270d84', u'system': u'linux_x86_64', u'version': u'1.10301.1'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stlink-linux_x86_64-1.10400.0.tar.gz', u'sha1': u'90c24db1d94b7555af7efff513effe60122cd302', u'system': u'linux_x86_64', u'version': u'1.10400.0'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stlink-windows-1.10200.1.tar.gz', u'sha1': u'79187839584ce0d585791ffb124af59c95c9bc34', u'system': [u'windows', u'windows_amd64', u'windows_x86'], u'version': u'1.10200.1'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-stlink-windows-1.10301.0.tar.gz', u'sha1': u'dfc53c39d10254013def26a6822490e965915e23', u'system': [u'windows', u'windows_amd64', u'window [... cut ...] io/dl-packages/tool-teensy-linux_x86_64-1.145.0.tar.gz', u'sha1': u'41e02a34121ce5ecf5b9690d03df753a64b830e6', u'system': [u'linux_x86_64'], u'version': u'1.145.0'}, {u'url': u'https://dl.bintray.com/platformio/dl-packages/tool-teensy-linux_x86_64-1.21.0.tar.gz', u'sha1': u'4aa6be9960e327470edb0c1658e5a5298ecdb052', u'system': [u'linux_x86_64'], u'version': u'1.21.0'}, {u'url': uTraceback (most recent call last):
  File "<string>", line 1, in <module>
IOError: [Errno 2] No such file or directory

ivankravets, 2019-07-16:
I did a new build of extension for you. See attached in this comment. Please unpack it, open VSCode > View > Command Palette > Install from VSIX... 
Select this development build. Restart VSCode.

Does now PIO works?

[platformio-ide-1.8.0-beta.2.vsix.zip](https://github.com/platformio/platformio-vscode-ide/files/3397411/platformio-ide-1.8.0-beta.2.vsix.zip)

ABOSTM, 2019-07-16:
Yes it looks like working. Thanks
At least I can run 'pio home' command successfully !! 
Can I use this version to work with, or it was just for test purpose ?

ivankravets, 2019-07-16:
Yes, you can use. The final release of 1.8.0 is coming today. Just click on WATCH button on the top of this page and you will be notified about this release.

Thanks for the provided information! Fixed via https://github.com/platformio/platformio-vscode-ide/commit/a61aa89f1c0c878baff7c8e4e23da2a517d2662d

closed 2019-07-16 by ivankravets

ABOSTM, 2019-07-16:
Thanks a lot for your help

mxadrian, 2020-05-19:
Hi, for me works only unchecking the Http: Proxy Strict SSL
Controls whether the proxy server certificate should be verified against the list of supplied CAs.

[image]

commit 6c2654ed (2019-07-16): Fixed an issue when PlatformIO Core does not handle properly disabled "Proxy Strict SSL" // Resolve #837