platformio/platformio-vscode-ide #840: Keyword for installing the extension is incorrect

larsvliet, 2019-06-18:
When searching for "platformio-ide" in VSCode the exension is not found:

[image]

With a space instead of a hyphen it works:

[image]

Please update the instructions on the website.

ivankravets, 2019-06-18:
It seems like a bug of VSCode. Could you report to https://github.com/Microsoft/vscode/issues ? It worked before and `platformio-ide` is ID of extension. In any case, we will update our docs and web site.

ivankravets, 2019-06-18:
Fixed in https://platformio.org/platformio-ide

closed 2019-06-18 by ivankravets