platformio/platformio-vscode-ide #857: Debugger: Window main.dbgasm isn't reloaded after change in source code

mirao, 2019-06-25:
**Steps to reproduce:**

- Debug main application `main.S` (Press F5)
- Debugger pauses on the first instruction at entry point (`Temporary breakpoint 1, main () at src/main.S:35`)
- Disassembly => Switch to assembly
- Stop debugger (Shift + F5)
- Switch to `main.S` window
- Insert an instruction as the first one in program's entry point, e.g. `NOP`
- Debug again (F5) 
- Debugger pauses at `NOP`
- Disassembly => Switch to assembly

**Expected result:**

- window `main.dbgasm` is refreshed and you can see `NOP` there

**Expected result:**
- window `main.dbgasm` is not refreshed.

**Workaround:**
- close window `main.dbgasm`
- and click to "Disassembly => Switch to assembly" again
- it works but it's impractical to do it after any change in source

**Used SW:**
- PlatformIO, version 3.6.7
- VSCode 1.35.1
- Ubuntu 19.04

**Used HW:**
- STM32F407G-DISC1, CPU Cortex-M4

ivankravets, 2019-07-13:
Is it possible without recompilation?

mirao, 2019-07-13:
Not sure what do you mean. If I press `F5` after a change in code, it is always recompiled.