platformio/platformio-vscode-ide #864: Start%20PIO%20Home%20Server

MartinRa01, 2019-06-28:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.35.1%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Windows_NT%2C%2010.0.18362%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20Traceback%20(most%20recent%20call%20last)%3A%0D%0A%20%20File%20%22%3Cstring%3E%22%2C%20line%201%2C%20in%20%3Cmodule%3E%0D%0A%20%20File%20%22%3Cstring%3E%22%2C%20line%2037%2C%20in%20%3Cmodule%3E%0D%0AException%3A%20PIO%20Plus%3A%20Runtime%20error%0D%0A%0A%09at%20P.Promise%20(C%3A%5CUsers%5CMartin%20Radvansk%C3%BD.WORKSTATION%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A15270)%0A%09at%20ChildProcess.h%20(C%3A%5CUsers%5CMartin%20Radvansk%C3%BD.WORKSTATION%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A5904)%0A%09at%20ChildProcess.emit%20(events.js%3A182%3A13)%0A%09at%20ChildProcess.cp.emit%20(C%3A%5CUsers%5CMartin%20Radvansk%C3%BD.WORKSTATION%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Ccross-spawn%5Clib%5Cenoent.js%3A34%3A29)%0A%09at%20maybeClose%20(internal%2Fchild_process.js%3A961%3A16)%0A%09at%20Process.ChildProcess._handle.onexit%20(internal%2Fchild_process.js%3A248%3A5)%0A%60%60%60%0A

MartinRa01, 2019-06-28:
No PIO Home

ivankravets, 2019-07-13:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/425

closed 2019-07-13 by ivankravets