platformio/platformio-vscode-ide #868: Installation%20Manager

hongthanks, 2019-07-02:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-vscode-ide%2Fissues%0A%0A%23%20Configuration%0A%0AVSCode%3A%201.35.1%0APIO%20IDE%3A%20v1.7.1%0ASystem%3A%20Windows_NT%2C%2010.0.17134%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20PIP%20Core%3A%20If%20you%20have%20antivirus%2Ffirewall%2Fdefender%20software%20in%20a%20system%2C%20try%20to%20disable%20it%20for%20a%20while.%20%0A%20%20%20Retrying%20(Retry(total%3D4%2C%20connect%3DNone%2C%20read%3DNone%2C%20redirect%3DNone%2C%20status%3DNone))%20after%20connection%20broken%20by%20'ConnectTimeoutError(%3Cpip._vendor.urllib3.connection.VerifiedHTTPSConnection%20object%20at%200x00000000047627F0%3E%2C%20'Connection%20to%20pypi.org%20timed%20out.%20(connect%20timeout%3D15)')'%3A%20%2Fsimple%2Fplatformio%2F%0D%0A%20%20Retrying%20(Retry(total%3D3%2C%20connect%3DNone%2C%20read%3DNone%2C%20redirect%3DNone%2C%20status%3DNone))%20after%20connection%20broken%20by%20'ConnectTimeoutError(%3Cpip._vendor.urllib3.connection.VerifiedHTTPSConnection%20object%20at%200x0000000005E39278%3E%2C%20'Connection%20to%20pypi.org%20timed%20out.%20(connect%20timeout%3D15)')'%3A%20%2Fsimple%2Fplatformio%2F%0D%0A%20%20Retrying%20(Retry(total%3D2%2C%20connect%3DNone%2C%20read%3DNone%2C%20redirect%3DNone%2C%20status%3DNone))%20after%20connection%20broken%20by%20'ConnectTimeoutError(%3Cpip._vendor.urllib3.connection.VerifiedHTTPSConnection%20object%20at%200x0000000005E39390%3E%2C%20'Connection%20to%20pypi.org%20timed%20out.%20(connect%20timeout%3D15)')'%3A%20%2Fsimple%2Fplatformio%2F%0D%0A%20%20Retrying%20(Retry(total%3D1%2C%20connect%3DNone%2C%20read%3DNone%2C%20redirect%3DNone%2C%20status%3DNone))%20after%20connection%20broken%20by%20'ConnectTimeoutError(%3Cpip._vendor.urllib3.connection.VerifiedHTTPSConnection%20object%20at%200x0000000005E394A8%3E%2C%20'Connection%20to%20pypi.org%20timed%20out.%20(connect%20timeout%3D15)')'%3A%20%2Fsimple%2Fplatformio%2F%0D%0A%20%20Retrying%20(Retry(total%3D0%2C%20connect%3DNone%2C%20read%3DNone%2C%20redirect%3DNone%2C%20status%3DNone))%20after%20connection%20broken%20by%20'ConnectTimeoutError(%3Cpip._vendor.urllib3.connection.VerifiedHTTPSConnection%20object%20at%200x0000000005E395C0%3E%2C%20'Connection%20to%20pypi.org%20timed%20out.%20(connect%20timeout%3D15)')'%3A%20%2Fsimple%2Fplatformio%2F%0D%0A%20%20Could%20not%20find%20a%20version%20that%20satisfies%20the%20requirement%20platformio%20(from%20versions%3A%20)%0D%0ANo%20matching%20distribution%20found%20for%20platformio%0D%0A%0A%09at%20f.value.i.runCommand%20(C%3A%5CUsers%5Cjs0915.hong%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A26495)%0A%09at%20ChildProcess.h%20(C%3A%5CUsers%5Cjs0915.hong%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Cplatformio-node-helpers%5Cdist%5Cindex.js%3A1%3A5904)%0A%09at%20ChildProcess.emit%20(events.js%3A182%3A13)%0A%09at%20ChildProcess.cp.emit%20(C%3A%5CUsers%5Cjs0915.hong%5C.vscode%5Cextensions%5Cplatformio.platformio-ide-1.7.1%5Cnode_modules%5Ccross-spawn%5Clib%5Cenoent.js%3A34%3A29)%0A%09at%20maybeClose%20(internal%2Fchild_process.js%3A961%3A16)%0A%09at%20Process.ChildProcess._handle.onexit%20(internal%2Fchild_process.js%3A248%3A5)%0A%60%60%60%0A

pbolduc, 2019-07-10:
@hongthanks do you really want help or are you just spamming the issues? You provide no information other. Do not post url encode values. If I was the moderator, I could just close your issue. What **your** garbage above says

# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.35.1
PIO IDE: v1.7.1
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: PIP Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while. 
   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x00000000047627F0>, 'Connection to pypi.org timed out. (connect timeout=15)')': /simple/platformio/
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x0000000005E39278>, 'Connection to pypi.org timed out. (connect timeout=15)')': /simple/platformio/
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x0000000005E39390>, 'Connection to pypi.org timed out. (connect timeout=15)')': /simple/platformio/
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x0000000005E394A8>, 'Connection to pypi.org timed out. (connect timeout=15)')': /simple/platformio/
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x0000000005E395C0>, 'Connection to pypi.org timed out. (connect timeout=15)')': /simple/platformio/
  Could not find a version that satisfies the requirement platformio (from versions: )
No matching distribution found for platformio

	at f.value.i.runCommand (C:\Users\js0915.hong\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:26495)
	at ChildProcess.h (C:\Users\js0915.hong\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:5904)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (C:\Users\js0915.hong\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

ivankravets, 2019-07-13:
It seems like broken Python installation or internet conenction

>   Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ConnectTimeoutError(<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x0000000005E395C0>, 'Connection to pypi.org timed out. (connect timeout=15)')':

closed 2019-07-13 by ivankravets