platformio/platformio-vscode-ide #873: Start PIO Home Server

tjaoc, 2019-07-04:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.36.0
PIO IDE: v1.7.1
System: Darwin, 19.0.0, x64

# Exception
```
Error: 
	at P.Promise (/Users/tcruz/.vscode/extensions/platformio.platformio-ide-1.7.1/node_modules/platformio-node-helpers/dist/index.js:1:15270)
	at ChildProcess.h (/Users/tcruz/.vscode/extensions/platformio.platformio-ide-1.7.1/node_modules/platformio-node-helpers/dist/index.js:1:5904)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.EventEmitter.emit (domain.js:442:20)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Socket.EventEmitter.emit (domain.js:442:20)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```

sstaub, 2019-07-04:
Same problem.

ivankravets, 2019-07-13:
Could you provide more details? Does it work when you open PlatformIO IDE Termina land type
```
pio home
```

CmdrChaos, 2019-07-23:
getting this same issue on OS X Mojave 10.14.5.  Error occurs during installation of the PlatformIO Extensions.

ivankravets, 2019-07-23:
@CmdrChaos could you open PlatformIO IDE Terminal and type `pio home`. Do you see any errors?

CmdrChaos, 2019-07-23:
@ivankravets ,
Error: Could not find a version that satisfies the requirement '~2.34.190418' for your system 'darwin_x86_64'

pio was not in my path, but once I found it, the above error was generated.

ivankravets, 2019-07-23:
> ~2.34.190418

It means, that you have Python 3.4 in the system. Could you remove it?

CmdrChaos, 2019-07-24:
Hi @ivankravets ,
removing Python 3.4 solved the issue.  Appreciate the assist.

closed 2019-07-24 by ivankravets