platformio/platformio-vscode-ide #879: ESP32 upload progress now shown

eos1d3, 2019-07-09:
On my Mac I have two SSD and each has PlatformIO using VSCode. All setting and OS version are the same.

But one of them will show the firmware upload progress with percentage like this and the other never shows:
```
Compressed 1176976 bytes to 657423...
Writing at 0x00040000... (31 %)
```

So what is the problem and how to get the progress as shown above?

pfeerick, 2019-07-13:
Are you running PlatformIO v4 on both? It sounds like the one that isn't showing progress might be running v3 still. Try running `pio --version` from the terminal in vscode to check, and if the one that isn't showing progress is still running 3.x, you can update it with `pio upgrade`.

eos1d3, 2019-07-13:
Thank for your reply. I think that setup is corrupted. I uninstalled PlatformIO and re-installed it and still got the same problem.

Then I un-installed VSCode and also PlatformIO and finally reinstalled everything, the progress can be seen now.

I just can't explained what happened.

pfeerick, 2019-07-13:
No worries... just saw it in amongst all the automated issues ;) Hm... well, at least it's working now. I suppose this issue can get closed now as without a reproducible setup, it's a moot point now.

closed 2019-07-13 by eos1d3